zephyrproject-rtos/zephyr #66909: M_PI is not defined

yashi, 2023-12-23:
**Describe the bug**

`M_PI` is not available when compiling with picolibc, both SDK and the module. Eventhough picolibc [already define `_DEFAULT_SOURCE`](https://github.com/picolibc/picolibc/commit/07bae261ad01304a46dc454499b6c81dde8d554e) when building with Zephyr.
It works if you define `_DEFAULT_SOURCE`.

It also works if you select minimal libc.

**To Reproduce**
Steps to reproduce the behavior:
1. Appliy the attached patch
    ````
    patch -p1 < a.patch
    ````
3. Build it
    ````
    west build -b qemu_cortex_m3 zephyr/samples/hello_world
    ````

**Changes to hello_world/src/main.c**

````diff
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index 2758d75d3fc..f2b222db846 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -5,9 +5,10 @@
  */
 
 #include <stdio.h>
+#include <math.h>
 
 int main(void)
 {
-       printf("Hello World! %s\n", CONFIG_BOARD);
+       printf("Hello World! %f\n", M_PI);
        return 0;
 }
````

**Expected behavior**
Compiled without any error.

**Logs and console output**
```
/home/yashi/src/zephyr-project/zephyr/samples/hello_world/src/main.c: In function 'main':
/home/yashi/src/zephyr-project/zephyr/samples/hello_world/src/main.c:13:37: error: 'M_PI' undeclared (first use in this function)
   13 |         printf("Hello World! %f\n", M_PI);
      |                                     ^~~~
/home/yashi/src/zephyr-project/zephyr/samples/hello_world/src/main.c:13:37: note: each undeclared identifier is reported only once for each function it appears in
[36/124] Building C object zephyr/arch/arch/arm/core/cortex_m/CMakeFiles/arch__arm__core__cortex_m.dir/fault.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/yashi/src/zephyr-project/build
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.4
 - Commit SHA: zephyr-v3.5.0-3464-gbeb43bdf202

**Additional context**
- https://github.com/picolibc/picolibc/commit/07bae261ad01304a46dc454499b6c81dde8d554e
- https://docs.zephyrproject.org/latest/doxygen/html/math_8h.html#ae71449b1cc6e6250b91f539153a7a0d3
- https://github.com/picolibc/picolibc/issues/649

keith-packard, 2023-12-24:
M_PI is not part of the C standard, nor does Zephyr appear to include it in the list of Zephyr extensions. If you want it in Zephyr, you'll either need to ask for it using `-D_POSIX_C_SOURCE=200908L` or change the definition of the Zephyr C language interface.

keith-packard, 2025-06-28:
It looks like it might need to be -D_XOPEN_SOURCE=700 instead. The POSIX spec is vague on this point, and picolibc is pretty clearly lacking here as it doesn't define __XSI_VISIBLE when only _POSIX_C_SOURCE is specified. I *think* picolibc should make the appropriate XSI symbols visible when _XOPEN_SOURCE is not specified but _POSIX_C_SOURCE is specified.