zephyrproject-rtos/zephyr #66915: doc: Documentation issue in 'develop/getting_started/index'

mytechnotalent, 2023-12-23:
**Describe the bug**

I have attempted to build Zephyr within Kali Linux on an x64-based architecture as well as a Debian Raspberry Pi 5 OS as well as Windows x64.  I have attempted to build the blinky project and I get the exact error each time.

**Environment**
1)x64 Kali Linux
2)ARM64 RP5
3)x64 Windows

```
west build -p always -b rpi_pico_w samples/basic/blinky
-- west build: making build dir /home/kali/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/kali/zephyrproject/zephyr/samples/basic/blinky
-- CMake version: 3.27.9
-- Found Python3: /usr/bin/python3 (found suitable version "3.11.7", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/kali/.cache/zephyr
-- Zephyr version: 3.5.99 (/home/kali/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: rpi_pico_w
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.4 (/home/kali/zephyr-sdk-0.16.4)
-- Found toolchain: zephyr 0.16.4 (/home/kali/zephyr-sdk-0.16.4)
-- Found Dtc: /home/kali/zephyr-sdk-0.16.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/kali/zephyrproject/zephyr/boards/arm/rpi_pico/rpi_pico_w.dts
-- Generated zephyr.dts: /home/kali/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/kali/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/kali/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/kali/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/kali/zephyrproject/zephyr/boards/arm/rpi_pico/rpi_pico_w_defconfig'
Merged configuration '/home/kali/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/home/kali/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/kali/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/kali/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/kali/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done (3.8s)
-- Generating done (0.0s)
-- Build files have been written to: /home/kali/zephyrproject/zephyr/build
-- west build: building application
[1/131] Preparing syscall dependency handling

[7/131] Generating include/generated/version.h
-- Zephyr version: 3.5.99 (/home/kali/zephyrproject/zephyr), build: zephyr-v3.5.0-3464-gbeb43bdf202b
[8/131] Performing configure step for 'second_stage_bootloader'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/kali/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/kali/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /home/kali/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done (0.2s)
-- Generating done (0.0s)
-- Build files have been written to: /home/kali/zephyrproject/zephyr/build/bootloader
[9/131] Performing build step for 'second_stage_bootloader'
[1/2] Building ASM object CMakeFiles/boot_stage2.dir/home/kali/zephyrproject/modules/hal/rpi_pico/src/rp2_common/boot_stage2/boot2_w25q080.S.obj
[2/2] Linking ASM executable boot_stage2
[58/131] Building C object CMakeFiles/app.dir/src/main.c.obj
[... 41 lines cut ...]
  110 |         DT_PHA_BY_IDX(node_id, gpio_pha, idx, pin)
      |         ^~~~~~~~~~~~~
/home/kali/zephyrproject/zephyr/include/zephyr/drivers/gpio.h:332:24: note: in expansion of macro 'DT_GPIO_PIN_BY_IDX'
  332 |                 .pin = DT_GPIO_PIN_BY_IDX(node_id, prop, idx),                 \
      |                        ^~~~~~~~~~~~~~~~~~
/home/kali/zephyrproject/zephyr/include/zephyr/drivers/gpio.h:367:9: note: in expansion of macro 'GPIO_DT_SPEC_GET_BY_IDX'
  367 |         GPIO_DT_SPEC_GET_BY_IDX(node_id, prop, 0)
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/home/kali/zephyrproject/zephyr/samples/basic/blinky/src/main.c:21:40: note: in expansion of macro 'GPIO_DT_SPEC_GET'
   21 | static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);
      |                                        ^~~~~~~~~~~~~~~~
/home/kali/zephyrproject/zephyr/include/zephyr/devicetree.h:238:25: note: in expansion of macro 'DT_CAT'
  238 | #define DT_ALIAS(alias) DT_CAT(DT_N_ALIAS_, alias)
      |                         ^~~~~~
/home/kali/zephyrproject/zephyr/samples/basic/blinky/src/main.c:15:19: note: in expansion of macro 'DT_ALIAS'
   15 | #define LED0_NODE DT_ALIAS(led0)
      |                   ^~~~~~~~
/home/kali/zephyrproject/zephyr/samples/basic/blinky/src/main.c:21:57: note: in expansion of macro 'LED0_NODE'
   21 | static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);
      |                                                         ^~~~~~~~~
[67/131] Building C object zephyr/drivers/clock_control/CMakeFiles/drivers__clock_control.dir/clock_control_rpi_pico.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/kali/zephyrproject/zephyr/build

```

thedjnK, 2023-12-23:
This board does not have an LED on it, therefore blinky will not compile because of the lack of an LED. You need to add an overlay to designate an LED

mytechnotalent, 2023-12-24:
@thedjnK the PicoW does in-fact have an onboard LED on PIN25 so not sure what you mean.

thedjnK, 2023-12-24:
So it does

yonsch, 2023-12-24:
The Pico W does have an LED, but it's not connected to the RP2040 directly -it's connected to the CYW43 chip. So theoretically we can have blinky working, but only after we have a driver for the wifi chip.

mytechnotalent, 2023-12-24:
@yonsch with the wide-adoption of the PicoW particularly with BTLE it would make sense to accommodate this port.  The CYW43 is well documented and should not be a heavy lift.

yonsch, 2023-12-24:
@mytechnotalent There's a Zephyr specific driver in the making by the chip's vendor, I can't find it at the moment

mytechnotalent, 2023-12-24:
@yonsch I was able to spend a few hours and use the low-level functions to work with the PicoW and ESP32 Wroom and it works successfully.  

I am working on the Bluetooth LE Security Study Guide and I had to change a number of includes to add the `zephyr` in front of it.  I am however not able to resolve.
```
fatal error: settings/settings.h: No such file or directory
   13 | #include <settings/settings.h>
```

Any thoughts?  Thank you in advance.

mytechnotalent, 2023-12-28:
@yonsch do we have any idea when the PicoW port will be supported as you mentioned there is a driver in the making?  Thanks again for your help and a Happy New Year!  I am excited to be involved in the Zephyr Community!

yonsch, 2024-01-01:
@mytechnotalent 

> do we have any idea when the PicoW port will be supported as you mentioned there is a driver in the making?

Turns out it was merged, but still without SPI support, see here: https://github.com/zephyrproject-rtos/zephyr/pull/63721#issuecomment-1779430801

> Happy New Year! I am excited to be involved in the Zephyr Community!

Happy new year, welcome aboard :)

mytechnotalent, 2024-01-02:
Thank you so much @yonsch I appreciate that.  Is there a work around to get the Bluetooth examples working properly with the Pico W as I am unable to even get anything to compile.

Xiretza, 2024-05-11:
Since I haven't seen it mentioned here yet, this is blocked by #67092.

jerzydziewierz, 2026-10-27:
i'll try to deal with this.