zephyrproject-rtos/zephyr #66917: doc: Documentation issue in 'boards/arm/xiao_ble/doc/index'

rabarar, 2023-12-23:
**Describe the bug**

Instructions below fail to build any Zephyr Sidewalk Sample for the Xiao_BLE_Sense Board

<< Please describe the issue here >>
The Sidewalks SDK doesn't contain a west.yml file to properly setup a west project and manifest 

<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `boards/arm/xiao_ble/doc/index`
* Version: main
* SHA-1: fe9b23070ce17ff29345263f473db504d1fcc39a

thedjnK, 2023-12-23:
There is no "sidewalk" in zephyr, whatever it is you refer to, it is not a part of this repo, contact the author for help

closed 2023-12-23 by thedjnK

rabarar, 2023-12-24:
Sidewalk is Amazon AWS SDK that works with zephyr.Your documentation on your Zephyr wiki page is erroneous.Leave it at your own laziness…

thedjnK, 2023-12-24:
> Your documentation on your Zephyr wiki page is erroneous

**There is no "sidewalk" in zephyr, whatever it is you refer to, it is not a part of this repo, contact the author for help**