zephyrproject-rtos/zephyr #66921: arch: posix: _POSIX_C_SOURCE declaration conflicts with the rest of the system

cfriedt, 2023-12-24:
**Describe the bug**
The native port (aka `CONFIG_ARCH_POSIX`) globally defines a number of symbols reserved by IEEE-1003.1-2017 (and many earlier versions of the standard) such as `_POSIX_C_SOURCE`, `_XOPEN_SOURCE`, and `_XOPEN_SOURCE_EXTENDED` for all Zephyr builds on that architecture, even Zephyr builds which do not explicitly want or use `CONFIG_POSIX_API=y` or any POSIX [subprofiling options](https://docs.zephyrproject.org/latest/services/portability/posix/option_groups/index.html).

When platforms or drivers inject global CFLAGS in this way, there is a strong likelihood that those CFLAGS will conflict over time.

Additionally, it oversteps boundaries a fair bit by attempting to manage POSIX API in Zephyr.

Indeed, there was already a conflict since the POSIX API used the value `200809L` (required by the specification) and ARCH_POSIX used `200809`. This created a warning that was then promoted to error by twister, which broken the build.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `native_sim_64`
 - What have you tried to diagnose or workaround this issue? I tried building code and saw a redefinition error due to conflicting values
 - Is this a regression? Probably. I don't think it's been followed closely though.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to build things without `_POSIX_C_SOURCE=200809`
2. Watch all of the warnings promoted to errors fly by

**Expected behavior**
The application and POSIX API should have control over the enabling the POSIX API for the global Zephyr build.

If `ARCH_POSIX` truly must have `_POSIX_C_SOURCE` defined, it should be done in a way that does not conflict or leak into that of the rest of the Zephyr build.

For example, the `arch/posix` library code could be built separately using a process similar to what is done in the [`external_lib`](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/application_development/external_lib) sample app. That way, that conflicting preprocessor defines for the library do not poison those of the rest of the Zephyr application.
 
In fact, nearly that exact approach is already being done with `scripts/native_simulator/Makefile`.

**Impact**
With some certainty, it will conflict with formalizing POSIX configuration via IEEE-1003.1-2017 Standard POSIX feature text macros, as described in [RFC: POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) which has been part of the release plan for over a year at this point and has had significant effort put into it.

**Logs and console output**

See any of the erroneous previous build attempts of #66762

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK v0.16.4
 - Commit: 8cf199fb04a6a78ea67a588712f187969ed1818b

**Additional context**
Discovered / fixed in the context of #66762

cfriedt, 2023-12-28:
It seems that this issue is also not avoided by using `zephyr_library_compile_definitions()`. There was a hypothesis that this would limit compile definitions to the library being built, but as of now, it's quite easy to demonstrate broken on `main`.

The same issue exists if `zephyr_compile_definitions()` are used in both locations.
```diff
diff --git a/arch/posix/CMakeLists.txt b/arch/posix/CMakeLists.txt
index dc83cecc61..8b7bd8ffe6 100644
--- a/arch/posix/CMakeLists.txt
+++ b/arch/posix/CMakeLists.txt
@@ -133,7 +133,7 @@ if (CONFIG_GPROF)
   target_link_options(native_simulator INTERFACE "-pg")
 endif()
 
-zephyr_compile_definitions(_POSIX_C_SOURCE=200809 _XOPEN_SOURCE=600 _XOPEN_SOURCE_EXTENDED)
+zephyr_library_compile_definitions(_POSIX_C_SOURCE=200809 _XOPEN_SOURCE=600 _XOPEN_SOURCE_EXTENDED)
 
 if (CONFIG_NATIVE_APPLICATION)
   zephyr_ld_options(
diff --git a/lib/posix/CMakeLists.txt b/lib/posix/CMakeLists.txt
index c6b3a88aea..d1c2bc391f 100644
--- a/lib/posix/CMakeLists.txt
+++ b/lib/posix/CMakeLists.txt
@@ -35,6 +35,7 @@ if(CONFIG_POSIX_API OR CONFIG_PTHREAD_IPC OR CONFIG_POSIX_CLOCK OR
 endif()
 
 zephyr_library()
+zephyr_library_compile_definitions(_POSIX_C_SOURCE=200809L _XOPEN_SOURCE=700)
```

```shell
west build -p auto -b native_sim_64 -t run tests/posix/common/
```
[image]

aescolar, 2023-12-28:
> It seems that this issue is also not avoided by using `zephyr_library_compile_definitions()`. There was a hypothesis that this would limit compile definitions to the library being built, but as of now, it's quite easy to demonstrate broken on `main`.

`zephyr_library_compile_definitions()` works fine. You need to
a) Also change the line setting those macros for PICOLIBC
b) For the posix arch cmake, to set it when you are in the scope of the library, not the whole Zephyr scope/not when  `ZEPHYR_CURRENT_LIBRARY` is the whole Zephyr.
 
```diff
diff --git a/arch/posix/CMakeLists.txt b/arch/posix/CMakeLists.txt
index dc83cecc61..7001e54e75 100644
--- a/arch/posix/CMakeLists.txt
+++ b/arch/posix/CMakeLists.txt
@@ -133,8 +133,6 @@ if (CONFIG_GPROF)
   target_link_options(native_simulator INTERFACE "-pg")
 endif()
 
-zephyr_compile_definitions(_POSIX_C_SOURCE=200809 _XOPEN_SOURCE=600 _XOPEN_SOURCE_EXTENDED)
-
 if (CONFIG_NATIVE_APPLICATION)
   zephyr_ld_options(
     -ldl
diff --git a/arch/posix/core/CMakeLists.txt b/arch/posix/core/CMakeLists.txt
index 3cf83c9fdd..06640d0675 100644
--- a/arch/posix/core/CMakeLists.txt
+++ b/arch/posix/core/CMakeLists.txt
@@ -1,6 +1,7 @@
 # SPDX-License-Identifier: Apache-2.0
 
 zephyr_library()
+zephyr_library_compile_definitions(_POSIX_C_SOURCE=200809 _XOPEN_SOURCE=600 _XOPEN_SOURCE_EXTENDED)
 zephyr_library_compile_definitions(NO_POSIX_CHEATS)
 zephyr_library_sources(
        cpuhalt.c
diff --git a/lib/libc/picolibc/CMakeLists.txt b/lib/libc/picolibc/CMakeLists.txt
index 232e4ba8fc..5811a9cefa 100644
--- a/lib/libc/picolibc/CMakeLists.txt
+++ b/lib/libc/picolibc/CMakeLists.txt
@@ -12,7 +12,7 @@ if(NOT CONFIG_PICOLIBC_USE_MODULE)
   # Use picolibc provided with the toolchain
 
   zephyr_compile_options(--specs=picolibc.specs)
-  zephyr_compile_definitions(_POSIX_C_SOURCE=200809)
+  zephyr_library_compile_definitions(_POSIX_C_SOURCE=200809)
   zephyr_libc_link_libraries(--specs=picolibc.specs c -lgcc)
   if(CONFIG_PICOLIBC_IO_FLOAT)
     zephyr_compile_definitions(PICOLIBC_DOUBLE_PRINTF_SCANF)
diff --git a/lib/posix/CMakeLists.txt b/lib/posix/CMakeLists.txt
index c6b3a88aea..d1c2bc391f 100644
--- a/lib/posix/CMakeLists.txt
+++ b/lib/posix/CMakeLists.txt
@@ -35,6 +35,7 @@ if(CONFIG_POSIX_API OR CONFIG_PTHREAD_IPC OR CONFIG_POSIX_CLOCK OR
 endif()
 
 zephyr_library()
+zephyr_library_compile_definitions(_POSIX_C_SOURCE=200809L _XOPEN_SOURCE=700)
 add_subdirectory_ifdef(CONFIG_GETOPT getopt)
 zephyr_library_sources_ifdef(CONFIG_EVENTFD eventfd.c)
 zephyr_library_sources_ifdef(CONFIG_FNMATCH fnmatch.c)
```

aescolar, 2023-12-28:
Removing the regression label as this has been the case since 2017. So it is not something that was working that broke due to some recent commit.

And clarifying that the issue is that if somebody wants to set _POSIX_C_SOURCE to 200809L they find themselves forced to set it to _POSIX_C_SOURCE=200809. So the assessment would be at an annoyance level, as the worst case is that somebody has some source setting this macro, and they need to either undef _POSIX_C_SOURCE first, or not set it in their code, or set it to the same value.

cfriedt, 2024-01-23:
Just setting this to medium (it was high), because it's affecting every platform, causes violations of [rules A.4 and A.5](https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html) for just about every platform, and the release is coming up.

I would prefer to make this `priority: high` (and it was originally set to that 1 month ago).

Can we have a status update as to when this will be fixed?

aescolar, 2024-01-23:
> Can we have a status update as to when this will be fixed?

@cfriedt currently you are blocking both PRs that are aiming at fixing this.

cfriedt, 2024-01-23:
> > Can we have a status update as to when this will be fixed?
> 
> @cfriedt currently you are blocking both PRs that are aiming at fixing this. So if you unblock them it should be faster.

@aescolar - there are no linked PRs that seem to fix this issue. I have commented on https://github.com/zephyrproject-rtos/zephyr/pull/67040 and suggested to use local compile definitions instead of global compile definitions.

Is there another PR that should be linked to this issue?

closed 2024-01-26 by nashif