zephyrproject-rtos/zephyr #66923: posix: implement `alarm()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`alarm()` is required as part of `POSIX_SIGNALS ` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/alarm.html

zacchaeusliang, 2023-12-29:
Hey i'm looking into this ticket any hints on how to start?

cfriedt, 2023-12-30:
> Hey i'm looking into this ticket any hints on how to start? 

Hi @zacchaeusliang - this is a pretty interesting topic.

Are you on discord at all? If so, can you ping me?

PR #74436 (2024-06-27): posix: add stubs for signal.h functions that need process support
Since Zephyr itself does not currently support processes, but conformant applications should still be able to link, add stubs for the remaining POSIX functions in the `POSIX_SIGNALS` Option Group.

The `POSIX_SIGNALS` Option Group is required for PSE51, PSE52, PSE53, PSE54, and likely many other POSIX Subprofiles.

> [!NOTE]  
> There is a Coding Guidelines false positive about declaring the `raise()` function (from C89).

[Doc Preview](https://builds.zephyrproject.io/zephyr/pr/74436/docs/services/portability/posix/index.html)

 [... cut]
PR #76327 (2024-08-23): posix: add stubs for signal.h functions that need process support
This is a resubmission of #74436.

There seems to have been an oversight in #75244 in that these commits (belonging neither to #73978 nor #74096) were reverted. No justification was made for PR #74436 to be reverted, so presumably it was an accident.

Indeed, all testing passed with this PR in `main` as well as against the `v3.7.0` tag. Also ran through CI successfully with the expanded testing in #76377.

-----

Since Zephyr itself does not currently support processes, but conformant applications should still be able to link, a [... cut]