zephyrproject-rtos/zephyr #66927: posix: implement `signal()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`signal()` is required as part of `POSIX_SIGNALS ` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/signal.html

douglashuangg, 2024-04-06:
Hey, I'd like to get started on this issue. Any pointers as to how I should tackle this issue?

cfriedt, 2024-04-07:
> Hey, I'd like to get started on this issue. Any pointers as to how I should tackle this issue?

@douglashuangg - that's a tricky one.

https://pubs.opengroup.org/onlinepubs/007904875/functions/sigaction.html

> It was intended that the signal() function should be implementable as a library routine using sigaction().

So you could technically make a very simple wrapper for it.

The caveat is that Zephyr does not currently support processes

RamPingle-eaton, 2025-03-23:
@douglashuangg Hope you have resolved this issue. If so, I would appreciate it if you could share the PR link here for me to review. Thank you

adityachopra29, 2025-09-01:
Hi, can I try this issue?

ritesh006, 2025-09-13:
@cfriedt is this means progress on signal() depends on the completion of the sigaction() kernel-level work? is my understanding correct?

cfriedt, 2025-09-18:
@ritesh006 - technically, `signal()` should depend on a new underlying Zephyr API, since `signal()` is part of ANSI / ISO C and `sigaction()` is part of POSIX (and POSIX is optional in Zephyr).

Zantarian, 2026-01-02:
https://github.com/zephyrproject-rtos/zephyr/pull/95358

Solution was merged.

Pagwin2, 2026-01-15:
Ah, okay so current status of this is that the stub is setup and if anyone wants to make further progress they need to write an actually implementation got it, thanks for the reply