zephyrproject-rtos/zephyr #66933: posix: implement `feof()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`feof()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/feof.html

DanTGL, 2024-10-24:
> @DanTGL - we probably need some ZVFS implementation of this (in fdtable.h) and a feof() wrapper in the common libc.

Ok, so the feof function in the common libc just calls the ZVFS implementation directly.
How should the file descriptor be handled?

DanTGL, 2024-10-25:
Most standard libraries seem to define `FILE` as a struct containing the file descriptor, end-of-file indicator, and some other stuff.
`feof` just checks the end-of-file indicator, it doesn't modify it.

Isn't the standard library already expected to handle that?

Am I misunderstanding something?

cfriedt, 2024-10-25:
> Most standard libraries seem to define `FILE` as a struct containing the file descriptor, end-of-file indicator, and some other stuff.

integer file descriptor? Obtained via `open()` or `_open()` or something like that?

That means they use POSIX under the hood, which is the dependency we want to remove.

> `feof` just checks the end-of-file indicator, it doesn't modify it.

That's right. I/O operations should set the EOF indicator when the end of a file has been reached. Hard to say if there needs to be a separate error bit.

> Aren't standard libraries already expected to implement `feof`?

Yes. However, many embedded C libraries conflate POSIX and ISO C (even ANSI C in this case), or make the assumption that the underlying OS has a native POSIX API.

> Am I misunderstanding something?

Not really. It's incredibly common.

The key difference here is that POSIX is optional in Zephyr, so we need to remove it from the dependency graph via an abstraction. Hence the ZVFS stuff.

So Zephyr's POSIX implementation depends on ZVFS and the ANSI C `FILE *` operations need to depend on ZVFS as well. Much of those APIs need to be syscalls as well.

The `struct fd_entry *` should be considered as Zephyr's native equivalent of `FILE *`.

I hope that helps to clarify things.

closed 2025-07-15 by cfriedt