zephyrproject-rtos/zephyr #66934: posix: implement `ferror()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`ferror()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/ferror.html

thong-phn, 2025-03-16:
Hi @cfriedt, there're 7 C libraries, including arcmwdt, armstdc, common, iar, minimal, newlib and picolibc. 
`~/zephyrproject/zephyr/lib/libc/arcmwdt
~/zephyrproject/zephyr/lib/libc/armstdc
~/zephyrproject/zephyr/lib/libc/common
~/zephyrproject/zephyr/lib/libc/iar
~/zephyrproject/zephyr/lib/libc/minimal
~/zephyrproject/zephyr/lib/libc/newlib
~/zephyrproject/zephyr/lib/libc/picolibc`

For this implement, will it be in the common C library?

I'm new to Zephyr so please correct me if I'm wrong.

abrarnasirjaffari, 2025-03-27:
Hi @cfriedt, I'd like to work on this issue. Could you assign this issue to me? Thanks!