zephyrproject-rtos/zephyr #66936: posix: implement `fgetc()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`fgetc()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/fgetc.html

00thirdeye00, 2024-07-08:
hej @ycsin, I am new to zephyr and would like to work on this issue. can you please guide me and give me some information/references for my understanding.

ycsin, 2024-07-09:
> hej @ycsin, I am new to zephyr and would like to work on this issue. can you please guide me and give me some information/references for my understanding.

Hi there, thanks for the interest for this issue.

Currently there's a stale (and subsequently closed) PR (#68839) for this issue that you might want to take a look, but I didn't find any implementation in the PR, only tests, cc @whythefyouflying

You can probably get some inspiration from how the `fputc` is implemented by referring to:
- [`drivers/console/uart_console.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/console/uart_console.c)
- [`lib/libc/minimal/source/stdout/stdout_console.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/libc/minimal/source/stdout/stdout_console.c)

When the target `FILE` is a `stdout`/`stderr`, the `fputc` or `putchar` functions are basically implemented by polling out bytes to the UART. The `fgetc` & `getchar` should be doing the opposite - read a byte from the UART. You probably need to implement the following:
- A `console_in` function in the `uart_console.c` driver
- Install (`__stdin_hook_install`) the `console_in` as `zephyr_fgetc`
- Implement `fgetc` with `zephyr_fgetc`

Please also take a look at #66944 ([`getchar`](https://en.cppreference.com/w/c/io/getchar)), which is a subset of this PR and could be done together.

jaiiarora, 2024-07-21:
Hi @ycsin 

I would like to work on this issue. Can you help me with it?
I have created a draft PR to begin with. (https://github.com/zephyrproject-rtos/zephyr/pull/76142/commits/7f975b84bcd1d39220e1be5c3b8dfc8998b67339#)

Thank you.