zephyrproject-rtos/zephyr #66937: posix: implement `fgets()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`fgets()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/fgets.html

Sh0g0-1758, 2024-01-19:
@cfriedt, do you have a draft PR for this? If not I would like to work on this.

AndersonTsai1203, 2025-04-09:
Hi @cfriedt. I'm new to zephyr project and I'm interested in learning and contributing to the project. Is there any guidance or documentation that I can refer to so that I could help with this issue?