zephyrproject-rtos/zephyr #66939: posix: implement `fopen()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`fopen()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/fopen.html

cfriedt, 2023-12-25:
Implementing this should fix #66132

cc @carlescufi

RAJAGOPALAN-GANGADHARAN, 2024-05-10:
Hi @cfriedt , I would like to implement this as well (and bunch of other posix methods like fclose, foef, ferror, and modifying fwrite)  since I'm already working on implementing the fread. Thanks!

brosier01, 2024-11-02:
Hi @stephanosio, Have you already started coding? I'd like to help you by working on this problem. Thanks !

cfriedt, 2024-11-02:
> Hi @stephanosio, Have you already started coding? I'd like to help you by working on this problem. Thanks ! 

@brosier01 - feel free to get started. I think what we'll need is a `zvfs_fopen()` syscall. A libc `fopen()` call would just be a wrapper around the syscall.

closed 2025-07-15 by cfriedt