zephyrproject-rtos/zephyr #66940: posix: implement `fread()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`fread()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/fread.html

RAJAGOPALAN-GANGADHARAN, 2024-05-01:
Hi @ycsin and @cfriedt , I would like to give this a shot. Can you please assign me this issue? I'm thinking of using this pr - https://github.com/zephyrproject-rtos/zephyr/pull/68516 as a base. Thanks!

cfriedt, 2024-05-01:
@RAJAGOPALAN-GANGADHARAN - done!

However, something to keep in mind is that `fread()` is actually just a part of ISO C, all the way back to C89. That means, that the implementation of this function should likely go into the common libc, and tests should be added to the common libc test suite.

https://en.cppreference.com/w/c/io/fread

Additionally, a fairly fundamental problem needs to be solved for this to be done The Right Way (TM). Namely, we need a way to easily associate `FILE *` objects to (and from) integer file descriptors.

That would ideally be done in ZVFS (which is itself a kind of relocation of code from various libraries and subsystems to avoid dependency cycles at the API level.

RAJAGOPALAN-GANGADHARAN, 2024-05-04:
Thanks @cfriedt - Looking at how fwrite has been implemented in minimal libc currently, seems like it only supports write to stdout and stderr. No fs support seems to be added (the use of zephyr fs apis like fs_open etc - https://docs.zephyrproject.org/latest/services/file_system/index.html#c.fs_dir_t.), is this what is expected from the fread (read from stdin) as well? Also in the future will the support be expanded to fs operations for these apis? Thanks!

cfriedt, 2024-05-04:
@RAJAGOPALAN-GANGADHARAN  Currently stdin / stdout / stderr should work, but we do want to support operations on other files as well.

de-nordic, 2024-05-07:
> @RAJAGOPALAN-GANGADHARAN - done!
>
> However, something to keep in mind is that `fread()` is actually just a part of ISO C, all the way back to C89. That means, that the implementation of this function [...]

@cfriedt Is correct.

C standard is subset of POSIX standard (or POSIX standard is superset over C).
This can be seen when comparing `read` and `fread` functions using the OpenGroup documentation:
read -> https://pubs.opengroup.org/onlinepubs/9699919799/functions/read.html
fread -> https://pubs.opengroup.org/onlinepubs/9699919799/functions/fread.html

Look also at the C standard, draft, here https://www.open-std.org/jtc1/sc22/wg14/www/docs/n3096.pdf - you will notice that the fread is part of stdlib definition (Chapter 7).

Note also that C standard wins with POSIX when differs:
[image]

closed 2025-07-15 by cfriedt