zephyrproject-rtos/zephyr #66942: posix: implement `fscanf()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`fscanf()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/fscanf.html

Sh0g0-1758, 2024-01-19:
@cfriedt, are you working on this? If not I would like to work on this.

CiaranRoche1, 2024-09-29:
Hi, just wondering if there has been any progress made on any of these libc functions, or am I all good to try to implement some of these myself?

cfriedt, 2024-09-30:
@CiaranRoche1 - this might depend on the changes included in the (unfortunately reverted) posix_device_io or posix_fd_mgmt PRs.

Again, it might make more sense to look at fixing a bug for your first PR, since the workflow could take some practice.

https://github.com/zephyrproject-rtos/zephyr/issues?page=3&q=is%3Aopen+is%3Aissue+label%3Abug

studentpiyush, 2025-03-10:
@cfriedt is anyone working on this. Can i take it from here?

studentpiyush, 2025-03-10:
@cfriedt  If you allow me, Can i take from this?

cfriedt, 2025-03-10:
@studentpiyush - feel free to make a PR for minimal libc. I believe other C libraries already have this function. Note: this is not a POSIX function.

studentpiyush, 2025-03-10:
Okay, I’ll keep that in mind and try to fix the issue. Thanks for allowing me to contribute!

closed 2025-07-15 by cfriedt