zephyrproject-rtos/zephyr #66943: posix: implement `getc()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`getc()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/getc.html

cfriedt, 2024-01-04:
Is this not already in the minimal libc??

cfriedt, 2024-07-29:
> I would like to take this. It would be helpful if you can guide and share some reference about this issue.

Hi @00thirdeye00 - `getc(FILE *stream)` shouldn't be too difficult.

From a very high level, it should be as simple as 
1. lock the fdtable entry associated with `stream` (and verify that it is valid)
2. read a character and report it back, or `EOF` if there is an error.
3. unlock the fdtable entry

Do we still have `zvfs_read()` or was that reverted? With that, it would be trivial to implement.

00thirdeye00, 2024-07-30:
@cfriedt hej, thank you, since i am new to zephyr, it can be helpful if you can provide any references to files, implementation and flow

cfriedt, 2024-07-30:
@00thirdeye00 -

lib/os/fdtable.c <-- ftdable and some ZVFS functions
lib/c_lib/common/ <-- common libc components

https://docs.zephyrproject.org/latest/contribute/index.html

seyoungjeong, 2025-06-28:
@00thirdeye00 @cfriedt has anybody started to work on this issue? Or can I take this issue?

seyoungjeong, 2025-06-29:
- Guess the `getc()` implementation would go under libc/minimal. And the definition would be `int getc(FILE* stream );`
- It sounds like that `FILE` handler could be either file or UART?

seyoungjeong, 2025-07-03:
I'm thinking to design and implement `getc()` using the same pattern as `putc()`. The design patten used for `putc()` leverages hooks(`_stdout_hook` for output) to abstract platform-specific behaviors. Similarly, `getc()` can use a `_stdin_hook` mechanism for input.

stephanosio, 2025-07-15:
> This was originally added as a ticket under POSIX, but `getc()` is not a POSIX function. It is from C89.
> 
> https://en.cppreference.com/w/c/io/fgetc
> 
> The expectation from the POSIX API is that this function is implemented as part of whatever C library is in use.

Minimal libc, by definition, only exists to meet the needs of the Zephyr "kernel", as defined by the [Coding Guidelines Rule A.4](https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html#rule-a-4-c-standard-library-usage-restrictions-in-zephyr-kernel).

Note that the Zephyr "kernel" is defined as follows in the Coding Guidelines Rule A.4:

> * Kernel (kernel)
> * OS Library (lib/os)
> * Architecture Port (arch)
> * Logging Subsystem (subsys/logging)

POSIX API is not part of the Zephyr "kernel" and, by design, the Zephyr POSIX subsystem was never meant to be and should never be used with the minimal libc.

@ycsin @cfriedt Given that Picolibc, the default standard C library for Zephyr, and all the other standard C libraries supported by Zephyr implement the `getc` function, I am in favour of closing this issue.

cfriedt, 2025-07-15:
Works for me.

I think a lot of these issues dealing with libc-related features were unfortunately created by mistake.

closed 2025-07-15 by cfriedt