zephyrproject-rtos/zephyr #66945: posix: implement `gets()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`gets()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/gets.html

cfriedt, 2024-01-04:
This API call is deprecated / unsafe.

The man pages say never use it and to use fgets instead.

https://man7.org/linux/man-pages/man3/gets.3.html

It might be worth making a note that this call is simply not supported in the documentation.

Likely picolibc / newlib has it though... 🤔

Wouldn't mind a second opinion from @keith-packard or @stephanosio

keith-packard, 2024-01-04:
glibc follows c11/c17 standards and doesn't even define `gets` unless you build with an older C standard (gcc -std=c99 provides gets). Even in that case, the API is marked as deprecated and you get a warning at compile time. And another warning at link time. Even building with `_POSIX_C_SOURCE=200809L` leaves gets undefined without also using the older ISO C standard.

This seems like a reasonable plan to me. Applications that want `gets` newline semantics (which differ from `fgets`, of course), should use the new ISO C `gets_s` API instead.

Picolibc and newlib don't provide `gets_s`, which should probably get fixed.

closed 2025-07-15 by cfriedt