zephyrproject-rtos/zephyr #66949: posix: implement `scanf()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`scanf()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/scanf.html

CiaranRoche1, 2024-09-29:
Hi, just wondering if there has been any progress made on any of these libc functions, or am I all good to try to implement some of these myself?

cfriedt, 2024-09-30:
> Hi, just wondering if there has been any progress made on any of these libc functions, or am I all good to try to implement some of these myself?

@CiaranRoche1 - I think the minimal libc might need an implementation.

Ideally, we would have an abstraction that could be used by both libc and POSIX as part of ZVFS. There are likely some hooks for picolibc and newlib that are involved. Minimal libc might need a bit more.

@ycsin - was there some progress on unlocked file I/o?

CiaranRoche1, 2024-09-30:
@cfriedt - This sounds interesting! Right now I am looking for something to tackle as my first issue, so I might take a crack at implementing a new function in libc. 

However, I would love to be involved in any progress you want to make on making an abstraction for libc and POSIX. Keep me posted!

cfriedt, 2024-09-30:
@CiaranRoche1 - alternatively, for your first issue, it might be good to find an easy bug to tackle at the url below. Unfortunately, the `good first issue` label is not used very consistently in Zephyr

amanem1, 2025-03-15:
hey @cfriedt  , is their any update on this issue ?
Can we continue working on this ?

cfriedt, 2025-03-15:
If you would like to make a PR to the minimal libc, feel free. Other libcs have a functional `scanf()` already. Note: it is not a POSIX function.

jonathanramirez4479, 2025-04-01:
I'm currently looking at this issue and trying to familiarize myself with the codebase. Keep in mind, I'm trying to make my first open source contribution. Are we to implement scanf() in the minimal/include/stdio.h file?

closed 2025-07-15 by cfriedt