zephyrproject-rtos/zephyr #66950: posix: implement `setbuf()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`setbuf()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/setbuf.html

snufkinwa, 2025-03-25:
Hi, I was wondering if there has been any progress on implementing these libc functions. If not, would it be okay for me to take on some of them?

cfriedt, 2025-03-28:
@snufkinwa - please do!

snufkinwa, 2025-04-30:
Thanks for your patience — circling back to this after some time.
I’ve reviewed the issue in detail and wanted to clarify: 

- setvbuf() is not part of the POSIX standard — it's a C89 standard library function.
- The expectation is that it will be provided by whatever C library is in use (e.g., newlib, picolibc).
- Zephyr's native simulator uses the host system's libc (such as glibc), where `setvbuf()` is already available by default.
- Therefore, no implementation is needed in the POSIX API layer.

To avoid future confusion, the appropriate step is to update the documentation at `doc/services/portability/option_groups/index.rst` to clarify the source of `setvbuf()` and its expected availability depending on the libc in use.

closed 2025-07-15 by cfriedt