zephyrproject-rtos/zephyr #66951: posix: implement `setvbuf()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`setvbuf()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/setvbuf.html

whythefyouflying, 2024-02-10:
@cfriedt I would like to work on this issue if possible, I am new to Zephyr and would love some tips on how to get this issue done.

cfriedt, 2024-02-11:
@whythefyouflying, I would suggest designing the testsuite, based on the specification. Sort of like test driven development.

Judging by the spec, `setvbuf()` might depend on `fopen()`,`fclose()`, and possibly others. So perhaps it's best to start on a different feature.

closed 2025-07-15 by cfriedt