zephyrproject-rtos/zephyr #66954: posix: implement `vscanf()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`vscanf()` is required as part of `POSIX_DEVICE_IO` [Subprofiling Option Group](https://pubs.opengroup.org/onlinepubs/9699919799/xrat/V4_subprofiles.html).

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/vscanf.html

CiaranRoche1, 2024-09-29:
Hi, just wondering if there has been any progress made on any of these libc functions, or am I all good to try to implement some of these myself?

cfriedt, 2024-09-30:
@CiaranRoche1 - same status as in #66949 I would guess. Help is always appreciated 🙏

abrarnasirjaffari, 2025-03-24:
Hi @cfriedt, I'd like to work on implementing vscanf(). Could you assign this issue to me? Thanks!

closed 2025-07-15 by cfriedt