zephyrproject-rtos/zephyr #66972: posix: implement `posix_trace_timedgetnext_event()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`posix_trace_timedgetnext_event()` is required as part of `_POSIX_TIMEOUTS` Option Group.

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/posix_trace_timedgetnext_event.html

raghav9-97, 2025-03-20:
I am new to Zephyr and would like to take up on this issue if no one else is working on this currently, I am still trying to wrap my head around the POSIX trace standard but would need some to implement it. Need a heads up to continue on this one.

cfriedt, 2025-03-21:
POSIX tracing is removed completely in the 2024 standard. In the 2008 revision it was already marked as obsolescent.

I think it's fair to close this ticket.

closed 2025-03-21 by cfriedt