zephyrproject-rtos/zephyr #66975: posix: implement `getmsg()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`getmsg()` is required as part of `_XOPEN_STREAMS` Option Group.

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/getmsg.html

Sh0g0-1758, 2024-01-17:
hey @ycsin, I am new to Zephyr and would like to tackle this issue. I have been looking into the docs of Zephyr and will start hacking as soon as you assign me this issue.

Sh0g0-1758, 2024-01-17:
@cfriedt @ycsin, can someone please assign me the issue and also point me towards some resources.

cfriedt, 2024-01-17:
@Sh0g0-1758 - for the POSIX STREAMS API, I think we'll basically follow Linux' lead. It's very rarely used in practice, so it will be unimplemented.

That basically means:

* 1st commit: create a stub function that will return -1 and `-ENOSYS`. Add Konfig / CMake as appropriate
* 2nd commit: update the "header test" to ensure that an application can link with the function
* 3rd commit: update the docs, mark it as "supported" in that an application will link, but also add a link back to the "undefined behaviour" blob (with the obelisk symbol)

That should allow you to play around enough to figure out how to add new code in Zephyr,  run and maybe debug Zephyr tests with west, run testsuites with twister, and maybe how to build Zephyr docs.

If you have any questions, please feel free to ask in the `#posix` channel on Zephyr's Discord.

Good luck, and thanks for your help 👍

PR #68516 (2024-03-11): posix: implement getmsg() and getpmsg()
This is part of the See https://github.com/zephyrproject-rtos/zephyr/issues/51211 (RFC https://github.com/zephyrproject-rtos/zephyr/issues/51211).

`getmsg()` and `getpmsg()` are required as part of _XOPEN_STREAMS Option Group.

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/getmsg.html

Fixes #66975
Fixes #66976

I add precision on the documentation that the function is a placeholder and will fail with ENOSYS.
closed 2024-03-11 by aescolar