zephyrproject-rtos/zephyr #66978: posix: implement `putmsg()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`putmsg()` is required as part of `_XOPEN_STREAMS` Option Group.

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/putmsg.html

uddivert, 2024-01-15:
Hey, I would like to tackle this issue.

I am new to zephyr is there any information about how to get started, and possible approaches this, and testing my implementation?

closed 2024-01-31 by carlescufi

ycsin, 2024-01-31:
> Hey, I would like to tackle this issue.
> 
> I am new to zephyr is there any information about how to get started, and possible approaches this, and testing my implementation?

This reply is probably a little too late, some pointers for you in case you haven't read them already:
- [Zephyr Getting Started Guide](https://docs.zephyrproject.org/latest/develop/getting_started/index.html)
- Tests are written in the ztest framework, you can read more about it [here](https://docs.zephyrproject.org/latest/develop/test/ztest.html)
- We run tests with Twister in Zephyr, you can read more about it [here](https://docs.zephyrproject.org/latest/develop/test/twister.html)

You may also take a look at #67711, which closed this issue, since @AbhinavMir is also a first-time contributor, there might be some answers in the PR.