zephyrproject-rtos/zephyr #66979: posix: implement `putpmsg()`

ycsin, 2023-12-24:
This is part of the See [POSIX Roadmap for LTSv3](https://github.com/zephyrproject-rtos/zephyr/issues/51211) (RFC #51211).

`putpmsg()` is required as part of `_XOPEN_STREAMS` Option Group.

For more information, please refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/putpmsg.html

YashSarnobat, 2023-12-26:
I would love to give it a try! Can you please tell me how to get started and how should I approach this?

cfriedt, 2023-12-26:
> I would love to give it a try! Can you please tell me how to get started and how should I approach this?

Awesome! I'm not super familiar with the streams API. I think it would maybe make sense for networking though.

I would suggest first creating this header in `include/zephyr/posix`

https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/stropts.h.html

Then add a 1st commit for the header that adds only the necessary data structures, macros, and types defined for this function.

Then add 2nd commit to implement the function (leave it as a stub to start off with, so make the function return -1 and set errno to `ENOSYS`).

Then add a 3rd commit to tests. These mainly need to ensure conformity to the spec.

I would make a new directory for these tests under `tests/posix/streams`.

AbhinavMir, 2024-01-16:
Hi, since this was inactive for 3 weeks, I picked up some work here. @ycsin I am a first time contributor, does this header file look right? I ported it over from the open group standard.

https://github.com/zephyrproject-rtos/zephyr/pull/67645

PR #73964 (2024-06-13): posix: streams: add stub for missing putpmsg() function
The putpmsg() function was left unimplemented when other functions in the `_XOPEN_STREAMS` Option were added.

Add a stub to complete the Option.

Only define `_XOPEN_STREAMS` if the Kconfig option is selected (rather than setting it to -1)

Additionally, clean up xopen streams doc to be a bit more presentable.

Fixes #66979

[Doc Preview](https://builds.zephyrproject.io/zephyr/pr/73964/docs/services/portability/posix/index.html)