zephyrproject-rtos/zephyr #66986: rpi_4b port does not build

mytechnotalent, 2023-12-24:
**Describe the bug**
When I try to build for the `rpi_4b` port I am getting a `zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c:21:10: fatal error: soc.h: No such file or directory` error.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `Windows 11 Home`
 - What have you tried to diagnose or workaround this issue? `attempted to locate soc.h`

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p always -b rpi_4brp`
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

thedjnK, 2023-12-26:
@lenghonglin

lenghonglin, 2023-12-27:
the board is `rpi_4b`,  not `rpi_4brp`

henrikbrixandersen, 2024-01-02:
> **Describe the bug** When I try to build for the `rpi_4b` port I am getting a `zephyrproject/zephyr/subsys/bluetooth/host/hci_core.c:21:10: fatal error: soc.h: No such file or directory` error.
...
>     1. `west build -p always -b rpi_4brp`

You neglected to specify which sample application/test you were trying to build and didn't provide the full output of the failing command - but from the above it seem you are trying to build a Bluetooth sample? Native Bluetooth on the RPi 4B is not yet supported by Zephyr, as far as I can tell (see https://docs.zephyrproject.org/latest/boards/arm64/rpi_4b/doc/index.html).

>     * Commit SHA or Version used `master`

We do not have a `master` branch. Please specify either the SHA or version used.

mytechnotalent, 2024-01-02:
Thank you @henrikbrixandersen I can close since native BT is not supported.

closed 2024-01-02 by mytechnotalent