zephyrproject-rtos/zephyr #67009: Blinky and Hello world not working for M487

DinkarY, 2023-12-26:
Hello, 

I am new in zephyr.
I have "NuMaker-M487KM V1.0" board.
I am trying to run the "Blinky" and "HelloWorld" demos. 
I build Blinky using the command: "**west build -p always -b nuvoton_pfm_m487 samples\basic\blinky**". and build successfully without error.
I build Helloworld using command: "**west build -p always -b nuvoton_pfm_m487  samples\hello_world**". and build successfully without error.
Flash code using command: "**west.exe flash --hex-file E:\ZephyrPath\zephyrproject\zephyr\build\zephyr\zephyr.hex**". and it was successful.
However, I am not able to observe the blink LED or Hello World on the UART terminal.
Then I try to debug code for blinky and hello_world. Found that code stuck at "arch_system_halt".
After that, I check the backtrace command. its result is as follows attached screen for both demo.
[image]

Can anyone help me to solve this issue?

Thanks in advance.

Dinkar Patel

ndrs-pst, 2023-12-26:
@DinkarY Based on the backtrace results indicating an error code of 26, which corresponds to `K_ERR_ARM_BUS_IMPRECISE_DATA_BUS `([LINK](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/arm/core/cortex_m/fault.c#L425-L428C3)).
Unfortunately, I don't have access to the specific target hardware in question. Therefore, my ability to assist is limited to compiling both the 'blinky' and 'hello_world' projects using SDK version 0.16.4 at the Zephyr commit-id: b702c8e8f01f09cb73ef554b129db0773f9f98a5.

I have done this with the files attached to this message. This should help determine if the issue is related to the software environment or not. [zephyr_samples_nuvoton_pfm_m487_2023-12-26.zip](https://github.com/zephyrproject-rtos/zephyr/files/13772689/zephyr_samples_nuvoton_pfm_m487_2023-12-26.zip)

keith-packard, 2023-12-26:
In this configuration, Zephyr assigns all ram from _end onwards to the malloc heap. I'm wondering if there is some additional hardware configuration required to enable all of that memory? It looks like this chip has three distinct memory areas, but I haven't dug through the docs in enough detail to understand if perhaps some of the memory is not enabled at power up and needs additional clock configuration?

You can try setting the `COMMON_LIB_MALLOC_ARENA_SIZE` variable to some small fixed amount (like 2048?) and see if that makes the examples at least start up. These examples don't actually use malloc, so you could also simply disable malloc entirely by setting `COMMON_LIBC_MALLOC` to `n`.

The other thing that confuses me a bit is whether the MPU is already enabled when `malloc_prepare` is called and whether the `z_malloc_partition` settings in that function need to be followed by some MPU register updates? Maybe @andyross or @npitre have some ideas here?

ndrs-pst, 2023-12-27:
@keith-packard Based on your analysis, you seem to be correct.
[image]

The sharing of SRAM2 with SPIM and insights from the Nuvoton HAL's startup code do indicate this.
https://github.com/zephyrproject-rtos/hal_nuvoton/blob/master/m48x/Devices/M480/Source/ARM/startup_M480.s#L214-L216

@DinkarY For a quick verification, you could either reduce the SRAM size from 0x28000 to 0x20000 by using overlay or enable CCMEN during startup.

andyross, 2023-12-27:
Saw this come by.  Yeah, this seems like a broken linkage setup for this board.  FWIW, building hello_world manually and looking at build/zephyr/linker.cmd, you can see that the "RAM" section is set to map 160kb at 0x20000000, which overlaps bank2, which starts at 128kb.  Linkage for cortex-m devices gets complicated, I'm pretty sure that value is coming out of CONFIG_SRAM_SIZE, but there are a lot variants.

You could try setting `CONFIG_SRAM_SIZE=128` in your prj.conf, and if that works submit to the board config somewhere with a note explaining that the third SRAM bank can't be used without proper SOC-layer integration to enable it.

DinkarY, 2023-12-27:
> Saw this come by. Yeah, this seems like a broken linkage setup for this board. FWIW, building hello_world manually and looking at build/zephyr/linker.cmd, you can see that the "RAM" section is set [...]

Thanks for the update. It works for me.

closed 2024-03-12 by nashif