zephyrproject-rtos/zephyr #67010: [Bluetooth][Mesh]: How to make sure the whole procedure completely while sending the ext adv?

TimChengMJ, 2023-12-26:
**Is your enhancement proposal related to a problem? Please describe.**
Q: While sending the mesh packets with using ext adv, it must need to check the "ADV_FLAG_ACTIVE" status. 
If the previous advertising termination is not reported or "BT_HCI_EVT_LE_ADV_SET_TERMINATED" packet lost, it will stuck the whole procedure, and cannot send any mesh packets over one advertising instance.
(Actually, it is important to fix the lost packet over HCI. But if the HCI transport is based on socket or uart etc, we cannot be sure the packet can be passed completely with 100%)

**Describe the solution you'd like**
Once we lost the important packets for mesh stack, is it possible to correct the procedure or fix the advertising?

**Describe alternatives you've considered**
1, Try to add timeout callback for one advertising.
2, Before start a new instance of advertising, check the previous advertise or check the expired time?

**Additional context**
Subsys: sys/bluetooth/mesh/adv_ext.c
Related function:  send_pending_adv/adv_sent

LingaoM, 2024-01-03:
This should not be something that Mesh needs to consider. AFAIK the entire Host does not consider this issue, for example:https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/hci_core.c#L331

TimChengMJ, 2024-01-04:
> This should not be something that Mesh needs to consider. AFAIK the entire Host does not consider this issue, for example:https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/hci_core.c#L331

Once it happened, it may be a disaster, and Mesh cannot recover it. As you see within hci_core.c, it only cause one error command , not entire Host.

PavelVPV, 2024-01-08:
Hi @TimChengMJ, thanks for the report! Were you able to reproduce this issue or is this hypothetical situation?

TimChengMJ, 2024-01-16:
> Hi @TimChengMJ, thanks for the report! Were you able to reproduce this issue or is this hypothetical situation?

You can try to mark down "BT_HCI_EVT_LE_ADV_SET_TERMINATED" report.(or lost once per 100times, and then the mesh stack will stuck, and cannot send any message.)

I can see this issue while packet lost with using hci-socket/hci-uart. i cannot be sure the packet can be passed completely with 100% for my real project.

PavelVPV, 2024-01-16:
I think I understood what you mean. I've tried running hci_uart sample on nRF52840 and zephyr app with ble host built as native_posix, but couldn't reproduce this issue so far. But I think I agree that this can happen. What is your setup/hw?

PavelVPV, 2024-01-16:
In any case, we will see how we can fix this.