zephyrproject-rtos/zephyr #67011: samples/bluetooth/peripheral on efr32xg24_dk2601b throw error: bt_settings: settings_subsys_init failed (err -45)

HonestQiao, 2023-12-26:
**Describe the bug**
I have successfully run samples/bluetooth/observer, but running samples/bluetooth/peripheral gives an error, indicating that the settings_subsys_init  call has an error, with an error code of -45.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p=always -b efr32xg24_dk2601b samples/bluetooth/peripheral`
2. `west flash`
3. use picocom connect to the board
4. See error

**Expected behavior**
Be able to successfully run samples/bluetooth/peripheral and provide the functions expected by the program

**Logs and console output**
```
 <inf> flash_gecko: Device flash-controller@50030000 initialized
*** Booting Zephyr OS build zephyr-v3.5.0-3466-gb702c8e8f01f ***
[00:00:00.000,000] <err> bt_settings: settings_subsys_init failed (err -45)
Bluetooth init failed (err -45)
```

**Environment (please complete the following information):**
 - OS: Ubuntu22.04
 - SDK: Zephyr SDK
 - board: xG24-DK2601B Silicon Labs
 - zephyr: commit b702c8e8f01f09cb73ef554b129db0773f9f98a5 (HEAD -> main, origin/main, origin/HEAD)
 - modules/hal/silabs: 7a6ba7c

**Additional context**
I tested samples/bluetooth/observer and it works fine.
`west build -p=always -b efr32xg24_dk2601b samples/bluetooth/observer`
`west flash`

output:
```
* Booting Zephyr OS build zephyr-v3.5.0-3466-gb702c8e8f01f ***
Starting Observer Demo
Started scanning...
Exiting main thread.
Device found: DE:CD:2F:AD:F3:F2 (public) (RSSI -72), type 0, AD data len 28
Device found: D0:B4:5D:53:EC:DE (public) (RSSI -71), type 0, AD data len 22
Device found: A4:C1:38:C1:2B:48 (public) (RSSI -99), type 0, AD data len 21
```

jhedberg, 2023-12-27:
I don't think this is Bluetooth related, rather something to do with the support for non-volatile storage on this specific board. Error number `45` is `EDEADLK`, and the only storage-related code I managed to find using that error is this:
https://github.com/zephyrproject-rtos/zephyr/blob/2312897055d19ce11ddfa496052e400afb8893f4/subsys/fs/nvs/nvs.c#L739-L743

Laczen, 2023-12-29:
@HonestQiao, this error typically appears when trying to use nvs storage on a area that is not erased before the first use. Erase the storage area and retry,

de-nordic, 2024-01-08:
Can you try erasing the entire flash of device and re-uploading the software?

fkokosinski, 2024-01-09:
Hi @HonestQiao! I've tried to reproduce this issue but wasn't able to with the instructions that you had provided:
```
*** Booting Zephyr OS build zephyr-v3.5.0-3838-g8ca40e5ebfb6 ***
[00:00:00.010,000] <inf> fs_nvs: 8 Sectors of 8192 bytes
[00:00:00.010,000] <inf> fs_nvs: alloc wra: 0, 1fe8
[00:00:00.010,000] <inf> fs_nvs: data wra: 0, 0
[00:00:00.020,000] <wrn> bt_hci_core: Controller to host flow control not supported
[00:00:00.021,000] <wrn> bt_ecc: ECC HCI commands not available
[00:00:00.021,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.023,000] <inf> bt_hci_core: Identity: 58:8E:81:63:09:5A (public)
[00:00:00.023,000] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x02ff
[00:00:00.023,000] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0x0000
Advertising successfully started
Indicate VND attr 0x802ff78 (UUID 12345678-1234-5678-1234-56789abcdef1)
```

closed 2024-03-12 by nashif