zephyrproject-rtos/zephyr #67021: RFC: Multi-subsystem / full feature samples

Thalley, 2023-12-27:
## Introduction

In Zephyr we have the [samples directory](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples) which is split into multiple sub-directories, often based on subsystems or specific functionality, like `bluetooth`, `drivers`, `sensor`, etc. This structure does not immediately provide any nice way to add (larger) samples (or demo applications) that span across multiple subsystems. 

### Problem description

I want to add one or more samples that span across multiple subsystems. One such example is a headset which would use a combination of sensors, drivers, bluetooth, USB, audio subsystem, etc. 

### Proposed change

The proposal could one of the following:

1. Add a new top-layer directory called `applications` or similar (similar to https://github.com/nrfconnect/sdk-nrf/tree/main/applications), which contains Zephyr based applications that go above and beyond what the smaller samples do. 
2. Add a new repository (similar to https://github.com/Azure/azure-sphere-gallery) that contain these example/reference applications

Or the solution could another one. 

## Detailed RFC

This RFC isn't more detailed than what has already been described. 

### Proposed change (Detailed)

This RFC isn't more detailed than what has already been described. 

### Dependencies

N/A

### Concerns and Unresolved Questions

Adding larger applications that span across multiple subsystems will require effort from multiple maintainers to keep up to date and runnable/buildable. Furthermore, the applications will likely have requirements and dependencies that may restrict them to only a small subset of boards that may require additional addons/shields to work. 

Another concern is that these applications may be looked at as reference designs, in which case it may take significant coordination between contributors to agree on how they should be designed. 

## Alternatives

The obvious alternative is of course to create a new repository outside of https://github.com/zephyrproject-rtos so as to disassociate the applications from the project itself, similar to how many other example applications have done (see https://github.com/golioth/awesome-zephyr-rtos for a list of some of these).

RalphCorderoy, 2024-01-11:
`applications` means a few too many things, especially to a newcomer looking for a reference design to study and modify into what they need so I'd favour `reference-design` or similar.

kartben, 2024-01-11:
> `applications` means a few too many things, especially to a newcomer looking for a reference design to study and modify into what they need so I'd favour `reference-design` or similar.

I need to formulate a more articulate reply (lost the reply I had started typing a few days ago before getting a chance to submit, sorry for the delay @Thalley) but just to follow up on this specifically:

Calling these "reference-designs" is IMO a really bold claim, and would have direct implications on how/where we plan on hosting these.
A "reference-design" would be, I think, expected to demonstrate best/references practices in every possible way we can think of, including how to best structure a *downstream* project building on top of Zephyr. In that regard, having the "apps" in the main Zephyr tree would not be ideal, and neither would having several apps in the same repo a la Sphere gallery, since I think we'd likely want to recommend an approach where the "app" is its own repo, with its own west.yml, etc. (T2 topology, a la [example-application](https://github.com/zephyrproject-rtos/example-application))

Thalley, 2024-01-12:
> Calling these "reference-designs" is IMO a really bold claim, and would have direct implications on how/where we plan on hosting these.

Agreed, that is a dangerous road to go on. 

> having the "apps" in the main Zephyr tree would not be ideal, and neither would having several apps in the same repo a la Sphere gallery, since I think we'd likely want to recommend an approach [...]

That's an approach I had not considered myself, but certainly makes sense. 
The downside of this is of course that the more repositories we use, the harder it is to ensure that they are properly updated, and the time/effort to maintain these increases.

kartben, 2024-05-16:
@Thalley #72856 just got me thinking about this one again. Do you think we could get some kind of Bluetooth headset "real" app in time for 3.7? I know this is getting late already, but on the other hand I am quite certain you guys probably have such a demo app handy and bringing it upstream might not be all that hard (and ofc I am willing to help)

jori-nordic, 2024-05-16:
What about going the other way, and having the main zephyr.yml pull in the `zephyr-applications` repo?
This way we can easily test them against PRs to the zephyr tree, keeping the code up-to-date.

henrikbrixandersen, 2026-09-16:
@kartben Is this still targeting v4.5?

kartben, 2026-09-16:
> [@kartben](https://github.com/kartben) Is this still targeting v4.5?

I'm afraid not, sorry.