zephyrproject-rtos/zephyr #67041: lib: unable to link to Zephyr's POSIX API when using a 3rd-party C library

cfriedt, 2023-12-28:
**Describe the bug**
It is currently not possible to link to Zephyr's POSIX API when using a 3rd-party C library.

Apparently, `CONFIG_EXTERNAL_LIBC` should be substituted with `CONFIG_HOST_LIBC` below. That way, `lib/posix` does not need to be excluded.

https://github.com/zephyrproject-rtos/zephyr/blob/87d056bd31b14b680e6f011134c7ce433c4462f6/lib/CMakeLists.txt#L7

Please also mention any information which could help others to understand
the problem you're facing:
 - **What target platform are you using?** Any / all
 - **What have you tried to diagnose or workaround this issue?** Encountered errors, gotten feedback from maintainers, and created a PR using the suggested fix.
 - Is this a regression? No

**To Reproduce**
Steps to reproduce the behavior:
1. apply the patch below
2. `west build -p auto -b qemu_riscv64 samples/hello_world`
3. See error

**Expected behavior**
I would expect to be able to include Zephyr's POSIX API. The error goes away with the linked PR.

**Impact**
It's a bit of a showstopper because I would like to use Zephyr's POSIX API.

**Logs and console output**
```shell
/samples/hello_world/src/main.c:17: undefined reference to `timespec_to_timeoutms'
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
```diff
diff --git a/samples/hello_world/CMakeLists.txt b/samples/hello_world/CMakeLists.txt
index ecb7d24bb8..d1423df02b 100644
--- a/samples/hello_world/CMakeLists.txt
+++ b/samples/hello_world/CMakeLists.txt
@@ -5,4 +5,4 @@ cmake_minimum_required(VERSION 3.20.0)
 find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
 project(hello_world)
 
-target_sources(app PRIVATE src/main.c)
+target_sources(app PRIVATE src/main.c src/libc.c)
diff --git a/samples/hello_world/prj.conf b/samples/hello_world/prj.conf
index b2a4ba5910..759313e918 100644
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1,2 @@
-# nothing here
+CONFIG_EXTERNAL_LIBC=y
+CONFIG_POSIX_CLOCK=y
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index 2758d75d3f..5ed15614b0 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -4,10 +4,16 @@
  * SPDX-License-Identifier: Apache-2.0
  */
 
+#include <errno.h>
+#include <stdint.h>
 #include <stdio.h>
+#include <time.h>
+
+extern int64_t timespec_to_timeoutms(const struct timespec *abstime);
 
 int main(void)
 {
-	printf("Hello World! %s\n", CONFIG_BOARD);
-	return 0;
+	errno = ECHILD;
+	perror("Hi!");
+	return timespec_to_timeoutms(NULL);
 }

```

closed 2023-12-28 by cfriedt

aescolar, 2023-12-28:
This bug report is incorrect in its description,
as explained here: https://github.com/zephyrproject-rtos/zephyr/pull/66998#discussion_r1436380262
The case provided in this bug report as an example of something which is prevented is not appropriate, as trying to do that, even if the POSIX_API was selectable would fail to build as there would be no C library selected.

cfriedt, 2023-12-28:
> This bug report is incorrect in its description,
> as explained here: https://github.com/zephyrproject-rtos/zephyr/pull/66998#discussion_r1436380262

This doesn't make any sense. This bug report and the attached PR implementing the change **you suggested** is exactly what is described.

[image]

> The case provided in this bug report as an example of something which is prevented is not

People are free to compile C code. It's OK.

> appropriate, as trying to do that, even if the POSIX_API was selectable would fail to build as there would be no C library selected.

No - there is no libc to select because ... it's external. That's the entire point. It's external 😏

aescolar, 2023-12-28:
> > This bug report is incorrect in its description,
> > as explained here: [#66998 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/66998#discussion_r1436380262)
> 
> This doesn't make any sense. This bug report and the attached PR implementing the change **you suggested** is exactly what is described.
> 
> ![Screenshot_20231228-092727.png](https://private-user-images.githubusercontent.com/983494/293223542-06bbf14b-317a-45d6-b9e3-8885429b72e7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM3NzQ1NzMsIm5iZiI6MTcwMzc3NDI3MywicGF0aCI6Ii85ODM0OTQvMjkzMjIzNTQyLTA2YmJmMTRiLTMxN2EtNDVkNi1iOWUzLTg4ODU0MjliNzJlNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIyOFQxNDM3NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNTIwNWYyYjkzMjc2NDNiMzRhODAwNGE1ODdiZDdjOGZjMTVlY2Q4MTVlOTczOWYxOTQ3Yzc5ODJkYmJiYWM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.8k_JE4dpAtKqUnTqZqDNAPMNHpydSlNNBjRcYhNj6xU)
> 
> > The case provided in this bug report as an example of something which is prevented is not
> 
> People are free to compile C code. It's OK.
> 
> > appropriate, as trying to do that, even if the POSIX_API was selectable would fail to build as there would be no C library selected.
> 
> No - there is no libc to select because ... it's external. That's the entire point. It's external 😏

The PR you sent is not what I suggested. The fixup PR I sent which you are blocking is.

cfriedt, 2023-12-28:
> The PR you sent is not what I suggested. The fixup PR I sent which you are blocking is.

Thank you for clarifying

PR #67042 (2023-12-28): lib: support linking to POSIX API with 3rd-party libc
Previously it was not possible to link to Zephyr's POSIX API under lib/posix when building with
`CONFIG_EXTERNAL_LIBC=y`.

This small change allows that to work as expected.

Fixes #67041