zephyrproject-rtos/zephyr #67050: LE Audio: Controller: Config settings with 44.1 kHz and 48 kHz are rejected at LE Create BIG

mkasenberg, 2023-12-28:
**Describe the bug**
Not high prio, posting just for the record:

Zephyr controller rejects the LE Create BIG if the fallowing config settings are used:
- 441_1
- 441_2
- 48_1
- 48_2
- 48_3
- 48_4
- 48_5
- 48_6

When the Zephyr controller is built with `CONFIG_BT_CTLR_CONN_ISO_LOW_LATENCY_POLICY=n` the LE Create BIG succeed, but only for the 441_1 setting. Even then the PTS device is not able to synchronize to the broadcast.

**To Reproduce**
With the AutoPTS and nRF5340 board.

**Impact**
Affected PTS test cases:
BAP/BSRC/STR/BV-09-C
BAP/BSRC/STR/BV-10-C
BAP/BSRC/STR/BV-26-C
BAP/BSRC/STR/BV-27-C
BAP/BSRC/STR/BV-28-C
BAP/BSRC/STR/BV-29-C
BAP/BSRC/STR/BV-30-C
BAP/BSRC/STR/BV-31-C
BAP/BSRC/STR/BV-32-C
BAP/BSRC/STR/BV-33-C

**Additional context**
@cvinayak

henrikbrixandersen, 2023-12-28:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Thalley, 2024-01-02:
We should add broadcast testing to the existing GMAP and CAP AC tests, which currently only covers unicast.

cvinayak, 2024-01-03:
@mkasenberg is there any pointer to be on "getting started" to be able to reproduce the failures reported? 

> To Reproduce
With the AutoPTS and nRF5340 board.

Assuming a new resource working on this issue, what are the needed steps to reproduce the issue?

Thalley, 2024-01-04:
> @mkasenberg is there any pointer to be on "getting started" to be able to reproduce the failures reported?
> 
> > To Reproduce
> > With the AutoPTS and nRF5340 board.
> 
> Assuming a new resource working on this issue, what are the needed steps to reproduce the issue?

I'll create a PR to add the CAP and GMAP broadcast audio configurations for BSIM

mkasenberg, 2024-01-04:
AutoPTS instruction can be found at:
https://docs.zephyrproject.org/latest/connectivity/bluetooth/autopts/autopts-win10.html
https://docs.zephyrproject.org/latest/connectivity/bluetooth/autopts/autopts-linux.html

Steps to reproduce:
1. Have nRF5340 board (revision 2.0.0 in my case)

2. Have 1 PTS dongle (the [LE Only one](https://support.bluetooth.com/hc/en-us/articles/360049018472-How-can-I-buy-the-PTS-dongle-))

3. Have installed the [PTS 8.5.3.4](https://www.bluetooth.com/develop-with-bluetooth/qualification-listing/qualification-test-tools/profile-tuning-suite/)

4. Have installed the [AutoPTS](https://github.com/auto-pts/auto-pts)
- [instructions for Windows users](https://docs.zephyrproject.org/latest/connectivity/bluetooth/autopts/autopts-win10.html)
- [instructions for Linux users](https://docs.zephyrproject.org/latest/connectivity/bluetooth/autopts/autopts-linux.html)

I use Windows as host, because the PTS starts faster than when I used Linux host + Windows virtual machine. But it should not matter. I have installed [Python 3.11.6.](https://www.python.org/ftp/python/3.11.6/python-3.11.6-amd64.exe).

5. Last Zephyr SHA that I tested and the issue still persisted is 3c0a6058b518f7d5dfbbadc9e857cc605632264b, so git-checkout on that:
```
git checkout 3c0a605
west update
```

6. Go to the zephyrproject/zephyr/samples/bluetooth/hci_ipc directory, then build and flash the app for the netcore:
```
west build -b nrf5340dk_nrf5340_cpunet -- -DOVERLAY_CONFIG='nrf5340_cpunet_iso-bt_ll_sw_split.conf;../../../tests/bluetooth/tester/nrf5340_hci_ipc_cpunet.conf'
west flash
```

7. Go to the zephyrproject/zephyr/tests/bluetooth/tester directory, then build and flash the app for the appcore:
```
west build -b nrf5340dk_nrf5340_cpuapp -- -DOVERLAY_CONFIG='nrf5340_hci_ipc.conf;overlay-le-audio.conf'
west flash
```

8. Close the PTS, if it is opened.

9. Go to the auto-pts directory and start the autopts server:
```
python autoptsserver.py
```

The autopts server will start the PTS in automation mode. Wait until the autoptsserver prints OK:
```
$ python autoptsserver.py
(2366879466768) Starting PTS 65000 ...
Local IP address: ('192.168.107.1', 'fe80::f87b:c093:ad28:dae8') DNS None
Local IP address: ('192.168.9.117', 'fe80::a694:4949:87af:e0b1') DNS None
Local IP address: ('192.168.58.1', 'fe80::f0b7:3be5:cc53:1889') DNS None
Serving on port 65000 ...
(2366879466768) OK
```

10. Find the jlink uart port name of the app core. There are usually two ports visible for nrf53. Find the one that prints these 5 bytes after board reset: `00 80 ff 00 00`

It is the BTP ready event. You can use the PuTTY or the below python script to check the port output:
```
import serial
com = 'COM4'
ser = serial.Serial(com, 115200, timeout=5)
ser.flushInput()
ser.flushOutput()
input('Reset board and press enter')
print(ser.read(9999))
ser.close()
```
There should be no other bytes in the output except those of the BTP Ready event.

11. Start the autopts client:
(You may have to adjust the port name and the IPs here)

- For Windows users:
```
python ./autoptsclient-zephyr.py zephyr-master -t COM4 -b nrf53 -c BAP/BSRC/STR/BV-09-C
```

- For Linux users:
```
python ./autoptsclient-zephyr.py zephyr-master -t /dev/ACM0 -b nrf53 -l 192.168.2.1 -i 192.168.2.2 -c BAP/BSRC/STR/BV-09-C
```
where -l is the IP of Linux host and -i is the IP of Windows guest on virtual machine.

Don't hesitate to ask questions under AutoPTS issues or you can reach me on discord, as the first launch can be troublesome.

Thalley, 2024-01-04:
@cvinayak https://github.com/zephyrproject-rtos/zephyr/pull/67210 adds testing of all audio configurations and low latency presets from the BAP spec. Quite a lot of them are not passing (see the commented out tests). 

This does not include the high reliability tests (yet).

cvinayak, 2024-01-09:
> @cvinayak #67210 adds testing of all audio configurations and low latency presets from the BAP spec. Quite a lot of them are not passing (see the commented out tests).
> 
> This does not include the high reliability tests (yet).

Work in progress addressing failures in #67210 is here: https://github.com/zephyrproject-rtos/zephyr/pull/67392

cvinayak, 2024-01-09:
@mkasenberg Thank you for your instructions, I will switch to reproducing the issues once I get to fix controller for the test in #67210

cvinayak, 2024-01-11:
@mkasenberg is it possible to run the failing PTS tests on a controller on this branch (before I get to setup my PTS)?  https://github.com/zephyrproject-rtos/zephyr/pull/67392

mkasenberg, 2024-01-11:
Yes, let's try if autopts bot is still operative.

cvinayak, 2024-01-12:
@mkasenberg Looks like PR run has failed, is the failures same as original failures or any improvements seen in the controller fro the PR?

mkasenberg, 2024-01-12:
I saw the bot logs and the patch works great, controller has successfully accepted the config settings. The autopts cron has reported fails, because an MMI handler in autopts has a bug and does not send anything, the fix is not merged yet https://github.com/auto-pts/auto-pts/pull/1067/commits/f1453575017757489ef8250dc57a3ca0c95b5181#diff-fc44e218eeb03508ffb17b3eda2ca77928b173334fe43d2c55ff23df09b09e46R315-R323

cvinayak, 2024-01-12:
thank you, added fixes comment for this gh issue in the Pr #67392 .... we can re-open this issue if there are problem when fixes are merged into main.

PR #67537 (2024-01-26): Bluetooth: Controller: Fix Broadcast ISO Create Scheduling
Fix Broadcast ISO Create Scheduling to consider reducing Host requested RTN value. And also use maximum overhead calculation when Extended Advertising and Periodic Advertising have not been started before BIG create.

Fixes #67050
closed 2024-01-26 by carlescufi