zephyrproject-rtos/zephyr #67058: RFC: Deeper code coverage profiling with Twister test automation

golowanow, 2023-12-28:
## Introduction

Twister tool [supports](https://docs.zephyrproject.org/latest/develop/test/coverage.html#coverage-reports-using-twister) dynamic code coverage profiling on test execution; it also generates human readable reports in various formats.
Currently this Twister feature provides less data than it is possible to achieve effectively on a CI pipeline 'at scale' for precise code profiling, test planning and gap analysis.

This proposal is to extend Twister in two aspects: 
* compose code coverage reports at deeper level per test suite instance build configuration.
* convey coverage results as an integral part of its test suite instance execution result.

### Problem description

* Twister currently process the GCOV coverage tool's output (or .gcno, .gcda files) and aggregates coverage statistics from all the test configurations which it executes for the test plan requested, so the final coverage report shows the **test plan** coverage summary, without details on how each **test instance configuration** built contributes to the statistics (unless the test plan is just one test configuration); worse if the test plan is for several platforms.
* moreover, there is no easy way to identify each test suite instance's contribution in the final report, e.g. to compare and find subtle profiling differences of each build configuration.
* In case of a test failure (e.g. because of occasional stack overflow by additional coverage data), or passed test with its coverage data garbled/truncated,  - these problems are hidden in the total numbers.
* The coverage data is a separate artifact and an additional effort is needed to bind it with the proper test execution data.

### Proposed change

1) Twister to generate separate coverage reports for each test suite configuration it executes.
2) Twister to include coverage data into `twister.json` as additional properties of its test suite run, right with the test results.

## Detailed RFC

1) The new `--coverage-split` mode extends Twister code coverage operations to report coverage statistics on each test instance configuration run individually in addition to the currently supported (default) reporting mode which aggregates data into a single
report summarizing results of all the tests in the current scope of the Twister run.
The split mode allows to identify precisely what amount of code coverage each test case configuration provides and to analyze its contribution to the overall test plan's coverage.
Each build directory will have its own coverage report and data files, so the overall disk space and the total execution time increase.
The split mode reads the raw data files while the test jobs are running in parallel, so the report aggregation stage doesn't need to do the same at its turn.
Another new `--disable-coverage-aggregation` option allows to execute only the `--coverage-split` mode when the summary statistics are not needed.

2) New Twister option `--coverage-sections` allows to select what coverage data to include into the `twister.json` file for its further analysis in addition to the test meta-data and the test results.
The choices are to select either the coverage summary, or the detailed branch coverage, ztest coverage, or all the coverage data collected.
Also the coverage run 'status' and coverage 'tool' values are added with any non-default choice. The 'environment' top object has additional 'gcov_tool' property to keep the command line argument selection.
Depending on `--coverage-split` and `--disable-coverage-aggregation` options the coverage data is attached either to its test suite object or/and at the report's top level as the aggregated summary of the test plan execution scope.

### Proposed change (Detailed)

see #66345 in review with the proposed Twister extension.

### Dependencies

The change should not affect any CI validation pipeline which uses Twister, unless the new reporting features are enabled explicitly.
With the new options enabled gradually (split aggregation, selected extra json sections) more data will be collected eventually increasing twister.json size and additional coverage reports.
The overall time for Twister to execute the same test plan on the same host(s) will also grow, although with better workload distribution - Twister will process GCOV output as parallel jobs instead of doing it once at the  final report generation, as currently.

### Concerns and Unresolved Questions

* The additional code coverage data in `twister.json` is fully supported only for the default `gcovr` reporting tool as it depends on gcovr capability. For `lcov` reporter the coverage run status is only added.
* The additional data will be an expected challenge to analyze, but isn't it a goal to have more data ?
* More complex json data. For example, the ElasticSearch needs to parse the extended multi-level `twister.json` sections;  to visualize it with Kibana it needs custom Vega charts. More details on this with examples to follow.

## Alternatives

* Use additional custom scripts or other tools to collect and aggregate GCOV artifacts differently after the Twister has done its part of the test plan execution.
* Split the test plan Twister execution into many small jobs for each test case configuration to execute it separately with its own code coverage reports.

golowanow, 2024-05-10:
@JordanYates, I've noticed you are working on twister + coverage improvements. 
Could you also have a look at this RFC and related #66345 ?

nashif, 2025-01-18:
> 1. Twister (optionally) to generate separate coverage reports for each test suite configuration it executes.

this is fine.

> 2. Twister (optionally) to include coverage data into twister_coverage.json as additional object properties of its test suite/instance together with the test results. It is better to keep this file separate from twister.json (although compatible schema) to avoid mixing a lot of coverage data with other test results; this is already done for memory footprint reports.

What is the purpose of saving this data and who is going to use it? How as a developer will I be able to use this data?

golowanow, 2025-01-18:
1. > What is the purpose of saving this data 

The purpose of saving this data is to 
> .. convey coverage results as an integral part of its test suite execution result.

2. > .. and who is going to use it? 

in this case it is rather for QA/validation/security audit people who use automation tools integrated into a data analysis pipeline in production cycle to store and analyze test results with corresponding code coverage aimed to detect execution path changes with its impact on their Zephyr-based product.

3. > How as a developer will I be able to use this data?

With whatever data analysis tool is available to that developer. In a simple personal case it might be  just python+pandas+scikit-learn (to name a few) to import a json file and explore/visualize it.

golowanow, 2025-01-30:
#66345 is reduced and resolves only part 1 of this RFC.