zephyrproject-rtos/zephyr #67060: blinky app build broken on adafruit_feather_nrf52840

StefJar, 2023-12-28:
**Describe the bug**
I am trying to use my good old adafruit feather nrf52840(prob 3y old). Its still supported

https://docs.zephyrproject.org/latest/boards/arm/adafruit_feather_nrf52840/doc/index.html

When building the sample blinky app it failing with:
```
(.venv) stefan@stefan-MS-7C37:~/build$ west build -b adafruit_feather_nrf52840 ~/zephyrproject/zephyr/samples/basic/blinky
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/stefan/zephyrproject/zephyr/samples/basic/blinky
-- CMake version: 3.27.4
-- Found Python3: /home/stefan/zephyrproject/.venv/bin/python3 (found suitable version "3.11.6", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/stefan/.cache/zephyr
-- Zephyr version: 3.5.0 (/home/stefan/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: adafruit_feather_nrf52840
-- Found host-tools: zephyr 0.16.4 (/home/stefan/zephyr-sdk-0.16.4)
-- Found toolchain: zephyr 0.16.4 (/home/stefan/zephyr-sdk-0.16.4)
-- Found Dtc: /home/stefan/zephyr-sdk-0.16.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/stefan/zephyrproject/zephyr/boards/arm/adafruit_feather_nrf52840/adafruit_feather_nrf52840.dts
-- Generated zephyr.dts: /home/stefan/build/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/stefan/build/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/stefan/build/build/zephyr/dts.cmake

warning: HAS_CMSIS_CORE (defined at modules/cmsis/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - CPU_CORTEX_M (defined at arch/arm/core/Kconfig:6), with value y, direct dependencies ARM (value: y), and select condition ARM (value: y)
Parsing /home/stefan/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/stefan/zephyrproject/zephyr/boards/arm/adafruit_feather_nrf52840/adafruit_feather_nrf52840_defconfig'
Merged configuration '/home/stefan/zephyrproject/zephyr/samples/basic/blinky/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at /home/stefan/zephyrproject/zephyr/cmake/modules/kconfig.cmake:355 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/stefan/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /home/stefan/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/stefan/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/home/stefan/zephyrproject/.venv/bin/python3 -B/home/stefan/build/build -GNinja -DBOARD=adafruit_feather_nrf52840 -S/home/stefan/zephyrproject/zephyr/samples/basic/blinky
```

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain zephyr-sdk-0.16.4

```
stefan@stefan-MS-7C37:~/zephyrproject/zephyr$ git status
HEAD detached at v3.5.0
nothing to commit, working tree clean

```

** Comment **

For me it sounds like

```
warning: HAS_CMSIS_CORE (defined at modules/cmsis/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - CPU_CORTEX_M (defined at arch/arm/core/Kconfig:6), with value y, direct dependencies ARM (value: y), and select condition ARM (value: y)
```

That the file

`adafruit_feather_nrf52840_defconfig`

selects the ARM MCU option

https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/adafruit_feather_nrf52840/adafruit_feather_nrf52840_defconfig#L8

and that `HAS_CMSIS_CORE` needs to be selected before or extra

thedjnK, 2023-12-29:
Builds fine, you have not got zephyr properly and are missing the cmsis module, ensure it was cloned properly by doing a `west update`, and (if using a custom manifest) ensure it is part of it

closed 2023-12-29 by thedjnK

StefJar, 2023-12-29:
Thx @thedjnK 
The `west update` solved that problem