zephyrproject-rtos/zephyr #67073: Timing is wrong when using target qemu_cortex_m3

koffes, 2023-12-29:
**Describe the bug**
When compiling for and running on qemu_cortex_m3, the kernel timings are off by a factor of >1000.

**To Reproduce**
1. use code similar to this:
```
#include <stdio.h>
#include <zephyr/kernel.h>

static uint32_t runtime;

int main(void)
{

	while (1) {
		k_msleep(1000);
		printk("Running %d\n", runtime);
		runtime++;
	}
	return 0;
}
```
2. `west build -p auto -b qemu_cortex_m4 samples/basic/blinky`
4. `west build -t run `

You will see that the print appears more than 1000 times a second.
By instead, building with `west build -p auto -b qemu_x86_64 samples/basic/blinky `
The timing seems just fine.

**Environment:**
 - OS: Ubuntu Linux  23.10, in WSL
 - Zephyr SDK toolchain (zephyr-sdk-0.16.4)
 - Commit SHA: Tested on aa4416624dceef29ec9ca78820c9fe5d3868f62c (current HEAD at time of writing).

gramsay0, 2024-01-07:
Some boards enable QEMU icount mode by default:
https://www.qemu.org/docs/master/devel/tcg-icount.html
https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_QEMU_ICOUNT

AFAIK this gives more deterministic execution but time differs from host wall-clock.

You could try disabling `CONFIG_QEMU_ICOUNT` or enabling `CONFIG_QEMU_ICOUNT_SLEEP` for more consistent guest/host time:
https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_QEMU_ICOUNT_SLEEP

microbuilder, 2024-01-08:
This is related to enabling ICOUNT, as mentioned by @gramsay0, and isn't really a bug in Zephyr's board per se:

- `arm_cortex-m3` has ICOUNT set via: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/qemu_cortex_m3/qemu_cortex_m3_defconfig#L10
- `qemu_x86_64` doesn't: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/x86/qemu_x86/qemu_x86_64_defconfig#L17

See: https://github.com/zephyrproject-rtos/zephyr/pull/58753#issuecomment-1573448605

In non-icount mode, the clock used by timer devices is the real-world wall clock, and guest execution is just "run as fast as possible". So a 1MHz timer will always fire after 1 microsecond real-world time (and if you program the timer to fire 'too fast' then you'll find the guest doesn't make forward progress because it's always handling timer interrupts).

In icount mode, the clock used by timer devices is tied to guest instruction execution, so timer expiry deadlines (and how far a free-running clock moves forward) are calculated as "after we've executed another N guest instructions".

Many boards require ICOUNT mode to be enabled for reliable execution, or you end up with timing issues on the emulated platform.

Closing this since it's not something we can really fix in Zephyr, but hopefully the above info gives some context to why you're seeing this behavior.

closed 2024-01-08 by microbuilder