zephyrproject-rtos/zephyr #67077: Implement Architecture-Specific Byte-Swap Instruction Support

ndrs-pst, 2023-12-30:
**Is your feature request related to a problem? Please describe.**
Lack of direct byte-swap instruction support in Zephyr RTOS hinders efficient data format conversions, especially in network communications and endian-sensitive applications.

**Describe the solution you'd like**
Implement built-in support for byte-swap instructions for __bswap_16 ... __bswap_64 in Zephyr, 
utilizing architecture-specific instructions (like BSWAP in x86, REV in ARM) where available, 
and falling back to software-based methods on other architectures.

**Describe alternatives you've considered**
The current approach is using software-based byte swapping or compiler-specific intrinsics, 
but these are less efficient and not as portable.

**Additional context**
Adding byte-swap instruction support would enhance Zephyr's performance in endian conversion tasks, 
benefiting IoT and network applications significantly.

henrikbrixandersen, 2023-12-30:
> benefiting IoT and network applications significantly.

Have you done any benchmarking on this proposal?

ndrs-pst, 2023-12-31:
@henrikbrixandersen I apologize for any overstatement with 'significantly,' as I haven't performed runtime performance benchmarking.
However, I can share insights from building **http_client** samples on the FRDM-K64F board, using various optimizations detailed in the table below. A rom_report is also attached for reference.

```
#include "cmsis_compiler.h"

#define __bswap_16(x)   ((uint16_t)__REV16((uint32_t)(x)))
#define __bswap_24(x)   ((uint32_t)(__REV(x) >> 8))
#define __bswap_32(x)   ((uint32_t)__REV(x))
#define __bswap_48(x)   ((uint64_t)(__bswap_64(x) >> 16))
#define __bswap_64(x)   (((uint64_t)__bswap_32((x) & 0xfffffffful) << 32) | (uint64_t)__bswap_32((x) >> 32))

```
| OPTIMIZATION  | FLASH SIZE (SW-BASED) | FLASH SIZE (BYTE-SWAP) | DIFF |
| ------------- | --------------------------- | ---------------------------------- | ---- |
| SIZE_OPTIMIZATION  | 178,180 B | 178,052 B | 128 B |
| SPEED_OPTIMIZATION  | 207,464 B | 207,300 B | 164 B |
| DEBUG_OPTIMIZATION  | 212,064 B | 210,484 B | 1,580 B |

[rom_report_samples_http_client_frdm_k64f_2023-12-31.zip](https://github.com/zephyrproject-rtos/zephyr/files/13801448/rom_report_samples_http_client_frdm_k64f_2023-12-31.zip)

jukkar, 2023-12-31:
Another endianess related PR #66920

andyross, 2024-01-01:
So, `__builtin_bswap_{16,32,64}()` are pervasively supported gcc extensions that do this in an optimal way at the toolchain level.  I guess the specific request here is for the API we expose in zephyr/sys/byteorder.h to use them?

ndrs-pst, 2024-01-01:
@andyross Yes, that's it. Using the `__builtin_bswap{16,32,64}()` functions seems like a viable option.
FWIW, the marginal reduction in code size from **SPEED_OPTIMIZATIONS** and **SIZE_OPTIMIZATIONS** may be due to the toolchain automatically replacing the swap macro with `__builtin_bswap`.
However, I personally tend to favor **DEBUG_OPTIMIZATIONS** until we're constrained by flash memory capacity since **SIZE_OPTIMIZATIONS** is really hard to debug. 😂

closed 2025-07-22 by ndrs-pst