zephyrproject-rtos/zephyr #67087: doc: Documentation issue in 'boards/arm/nucleo_g0b1re/doc/index'

WardBenjamin, 2024-01-01:
**Describe the bug**

The STM32G0B1 has two CAN-FD peripherals available. Zephyr already supports one since https://github.com/zephyrproject-rtos/zephyr/pull/54403. Here is the current permalink: https://github.com/zephyrproject-rtos/zephyr/blob/e67b12e63611291ac567bac4abfb9d43ecac0ce8/boards/arm/nucleo_g0b1re/nucleo_g0b1re.dts#L180 

Meanwhile, the "[supported features](https://docs.zephyrproject.org/latest/boards/arm/nucleo_g0b1re/doc/index.html#supported-features)" matrix on the Nucleo G0B1RE page does not list any CAN support. 

**Environment**

* Page: `boards/arm/nucleo_g0b1re/doc/index`
* Version: main
* SHA-1: 5d5249d85bb7bb779314693baa230195c0b070e0

WardBenjamin, 2024-01-01:
Side note: what needs to happen for Zephyr to properly support the second CAN-FD peripheral on this chip? I saw some notes about shared IRQs being an issue here https://github.com/zephyrproject-rtos/zephyr/pull/34963 but it is unclear to me whether that's been resolved in the two years since.

WardBenjamin, 2024-01-01:
It looks like this is also an issue on the [Nucleo G474 page](https://docs.zephyrproject.org/latest/boards/arm/nucleo_g474re/doc/index.html#supported-features): https://github.com/zephyrproject-rtos/zephyr/blob/e67b12e63611291ac567bac4abfb9d43ecac0ce8/boards/arm/nucleo_g474re/nucleo_g474re.dts#L207

henrikbrixandersen, 2024-01-10:
Documentation additions: #67430

PR #67430 (2024-01-11): boards: arm: nucleo_g0b1re, nucleo_g474re: list FDCAN1 as supported in the docs
List FDCAN1 as supported in the board documentation.
    
Fixes: #67087
commit 735c9e23 (2024-01-11): boards: arm: nucleo_g474re: list FDCAN1 as supported in the docs

List FDCAN1 as supported in the board documentation.

Fixes: #67087
commit fcef0607 (2024-01-11): boards: arm: nucleo_g0b1re: list FDCAN1 as supported in the docs

List FDCAN1 as supported in the board documentation.

Fixes: #67087
closed 2024-01-11 by henrikbrixandersen