zephyrproject-rtos/zephyr #67092: drivers: wifi: Infineon AIROC: Add SPI support

yonsch, 2024-01-01:
**Is your enhancement proposal related to a problem? Please describe.**
The Infineon AIROC Wifi driver currently only supports the SDIO interface. This makes it unusable for boards where the SPI interface is used, such as the Raspberry Pi Pico W.

**Describe the solution you'd like**
Add support for SPI.

**Additional context**
As mentioned [here](https://github.com/zephyrproject-rtos/zephyr/pull/63721#issuecomment-1779430801):

> While we have provided the infrastructure to allow for SPI support, we aren't yet working on enabling that. We have an internal backlog and SPI is not yet very high on it. We would be open to any assistance with SPI AIROC Wi-Fi support the community can provide.

Opened this issue to track progress on this task, and possibly find a volunteer to complete it.

ThreeEights, 2024-02-28:
@yonsch - The PIO SPI driver support for 3-wire mode, needed to use the CYW43439 chip on the Pico W, is now in final testing;  I expect to enter the pull request by the end of this week.  And I have the Infineon HAL layer pulled apart, trying to work in the SPI support.  Not sure when I can get that working.

SamsonBox, 2024-04-20:
Is there any plan, when this feature will be available? It would be great to use the AIROC Chip on the Pico W Board.

ThreeEights, 2024-04-23:
@SamsonBox - I'm working on it with help from Infineon.  I think it's a matter of a couple weeks to get it working reliably, followed by however long it takes to get it integrated into Zephyr.  (We have our own Pico W projects waiting for this!)

SamsonBox, 2024-04-23:
@ThreeEights Thank you for your answer. Sounds great!

eltiempopaso, 2024-08-25:
Is this already working? (Infineon driver thru spi, to be used in rpi Pico w)

eltiempopaso, 2024-08-26:
> > @ThreeEights Is this already working? (Infineon driver thru spi, to be used in rpi Pico w)
> 
> Alas, not yet. I'm still debugging.

Do we know when it'll be ready? I do not know whether there is any plan

Can I help you somehow? I'm interested in having the rpi pico wifi up&working.

ThreeEights, 2024-12-03:
@SamsonBox @eltiempopaso - If you're still interested in using WiFi on the Raspberry Pi Pico, my pull request is ready for people to test it:  https://github.com/zephyrproject-rtos/zephyr/pull/78773.  It's based on Zephyr's main branch (4.0 and later).  Building samples/net/wifi/shell works out of the box (at least for me!):
```
west build -b rpi_pico/rp2040/w
```
Please let me know if you run across any problems!

closed 2024-12-10 by kartben

eltiempopaso, 2024-12-10:
ok great! I am now focused on other things, i'll come back to you when I have the chance to test it.

thanks for letting me know!