zephyrproject-rtos/zephyr #67099: boards: ebyte_e73_tbb: dead link

nordicjm, 2024-01-02:
**Describe the bug**
The `ebyte_e73_tbb` board links to a website that no longer exists, the new company domain appears to be cdebyte.com

Whilst this should be fixed, this issue is more for @kartben to see if we have a way to check all the links that files in `boards/*/*/doc/*.rst` go to and see if they are still valid?

nordicjm, 2024-01-02:
`decawave_dwm1001_dev` too

kartben, 2024-01-02:
Thanks for reporting these, @nordicjm!

We actually have a Sphinx `ninja linkcheck` target to take care of finding dead links which I am running regularly. It is taking quite a long time to run, and may sometimes lead to false positives, so I am not sure it makes sense to add it in CI, but I'll look into it. Could be something that we run only once a week, maybe.

You're also bringing up the topic of what to do for boards that are not commercially available/supported anymore. While completely removing them from the source tree might be counter-productive (even if the board is EOL or the company behind it doesn't exist anymore, the port itself should still work), it might make sense to move such boards to an "attic" section of some sort?

nordicjm, 2024-01-02:
Running it weekly, or even monthly would be good.

Well for the `acn52832` board I just mean that the link itself is a 404 and that's with ignoring the invalid SSL certificate, was not meaning removal of that board, maybe there is a working documentation link somewhere but I did not search (was only looking for the company name). Boards that I _would_ advocate removal of would be things like `boards/arm64/bcm958402m2_a72` - there have been many issues with this, the chip has never been publicly available nor any documentation for it and the submitters have not responded at all to questions going back a year (if not longer), fully removing boards like that would be a good idea

commit 8f09325e (2024-03-15): doc: boards: Update E73-TBB product page link

E73-TBB board now lives at cdebyte.com.
Fixes #67099.
closed 2024-03-15 by nashif