zephyrproject-rtos/zephyr #67128: RFC: drivers: can: remove run-time RTR filtering, add build-time RTR filter

henrikbrixandersen, 2024-01-02:
## Introduction

The Controller Area Network (CAN) 2.0B specification allows two types of CAN frames; Data frames or Remote Transmission Request (RTR) frames. Data frames carry up to 8 bytes of data payload, whereas RTR frames carry no data payload but instead request the recipient of the frame to respond with a Data frame containing a specific number of data bytes.

### Problem description

Traditionally, CAN controllers have allowed for hardware-based filtering of incoming CAN frames based on the frame type, but a growing number of CAN controllers do not have support for individual RX hardware filters based on the Remote Transmission Request (RTR) bit. As the Zephyr CAN controller API allows for filtering on the frame type, this leads to various work-arounds on the Zephyr driver level mixing hardware and software filtering (which in turn leads to issues as described in #64554).
    
As the use of RTR frames is discouraged by CAN in Automation (see e.g. CiA application note 802 v1.1.0: "CAN remote frames: Avoiding of usage") - and not even supported by the newer CAN FD (and CAN XL) standards - this often leads to unnecessary overhead, added complexity, and worst-case to non-portable behavior between various CAN controller drivers.

### Proposed change

Instead, move to a simpler approach where the ability to accept/reject RTR frames is globally configured via Kconfig. By default, all incoming RTR frames are rejected at the driver level, a setting which can be supported in hardware by most in-tree CAN controllers drivers, either via individual RX filter options or a global RTR rejection option. Some drivers already rely solely on software filtering (e.g. drivers for CAN controllers with a very limited amount of RX filters available). For these drivers, this refactoring of RTR filtering still reduces code path complexity.

An overview of the currently supported CAN controllers along with their support for RTR filtering is shown in the table below.

| Driver compatible | Supports individual RTR filters | Supports global RTR reject/accept | Uses software filtering | Needs additional software RTR reject | 
| - | - | - | - | - |
| atmel,sam0-can | | X | | |
| atmel,sam-can | | X | | |
| espressif,esp32-twai | | | X | |
| kvaser,pcican | | | X | |
| microchip,mcp2515 | | | X | |
| microchip,mcp251xfd | | | | X |
| nuvoton,numaker-canfd | | X | | |
| nxp,flexcan-fd | X | | | |
| nxp,flexcan | X | | | |
| nxp,lpc-mcan | | X | | |
| nxp,s32-canxl | (X) | | | |
| renesas,rcar-can | | | (X) | |
| st,stm32-bxcan | X | | | |
| st,stm32-fdcan | | X | | |
| st,stm32h7-fdcan | | X | | |
| ti,tcan4x5x | | | X | |
| zephyr,can-loopback | | | X | |
| zephyr,native-linux-can | | | X | |

Legacy applications or protocol implementations where RTR reception isrequired, can select a new `CONFIG_CAN_ACCEPT_RTR` Kconfig option to accept incoming RTR frames matching added CAN filters. These applications or protocols will need to distinguish between RTR and data frames on their own in their respective CAN RX frame handling routines.

## Detailed RFC

Please see #67127

### Proposed change (Detailed)

Please see #67127

### Dependencies

This is a breaking API change in the CAN controller driver API, which is currently marked stable.

### Concerns and Unresolved Questions

The hardware support for RTR reception/filtering in the following drivers is a bit unclear to me (and I do not have the hardware to verify it):
- nxp,s32-canxl
- renesas,rcar-can

## Alternatives
* Leave the CAN controller API as-is and attempt to add/ proper software/driver workarounds to address the mismatch between the current API and the increasingly common hardware, where individual RTR filtering is not implemented.
* Allow CAN controller drivers to optionally support RTR filtering (which is one of the work-arounds currently used in use in-tree). This, however, leads to issues as described in https://github.com/zephyrproject-rtos/zephyr/pull/65058#pullrequestreview-1727200186
* Go with this solution, but negate `CONFIG_CAN_ACCEPT_RTR` and call it `CONFIG_CAN_REJECT_RTR` (selected by default) as this is likely easier to understand by end users. This, however, this leads to problems for protocol implementations wishing to `select` RTR support via Kconfig.

congnguyenhuu, 2024-01-03:
Hello @henrikbrixandersen, for NXP S32 CAN XL, although HW CAN XL supports individual RTR filters, but HAL CAN XL driver has limitation that not support send RTR frames (only support support send data frames). So current CAN XL shim driver not supports RTR filter and also should do this skip all RTR tests as not support RTR filter in this propsed change.

henrikbrixandersen, 2024-01-03:
> Hello @henrikbrixandersen, for NXP S32 CAN XL, although HW CAN XL supports individual RTR filters, but HAL CAN XL driver has limitation that not support send RTR frames (only support support send [...]

That is unfortunate. Is it possible to circumvent the HAL for this by direct register access in the shim driver?

martinjaeger, 2024-01-03:
Thanks @henrikbrixandersen for this very detailed RFC including the analysis of all currently supported CAN hardware.

As already discussed offline, I agree that this is the best approach to deal with RTR.

I'm confident that a solution for the CAN XL driver can be found.

congnguyenhuu, 2024-01-03:
> That is unfortunate. Is it possible to circumvent the HAL for this by direct register access in the shim driver?

It is not  possible, must update HAL drivers for supporting. I will investigate this.

simhein, 2024-01-08:
@henrikbrixandersen Thanks for the clear proposal +1 form my side for the solution you described.

congnguyenhuu, 2024-01-15:
> It is not possible, must update HAL drivers for supporting. I will investigate this.

I opened a PR update HAL to allow for supporting RTR frames https://github.com/zephyrproject-rtos/hal_nxp/pull/318

henrikbrixandersen, 2024-01-15:
> I opened a PR update HAL to allow for supporting RTR frames [zephyrproject-rtos/hal_nxp#318](https://github.com/zephyrproject-rtos/hal_nxp/pull/318)

Thanks! I've left a couple of questions there.

In order not to stall the switch from run-time RTR filters to build-time I have modified https://github.com/zephyrproject-rtos/zephyr/pull/67127 to skip RTR testing if the driver does not support transmission of RTR frames.

congnguyenhuu, 2024-01-16:
> Thanks! I've left a couple of questions there.
>
> In order not to stall the switch from run-time RTR filters to build-time I have modified #67127 to skip RTR testing if the driver does not support [...]

I agree this. I will to do for support later

closed 2024-01-21 by henrikbrixandersen