zephyrproject-rtos/zephyr #67138: Bluetooth: L2CAP: single TX path for dynamic and static channels

jori-nordic, 2024-01-03:
In the L2CAP module, there is are currently two very different codepaths that depend on the channel type.

This is problematic, as the functions are named similarly, increasing cognitive load when trying to follow code.
It's also a problem when reasoning about ATT behavior, as EATT uses dynamic chans where the UATT channel uses the static ATT channel. A GATT operation can go over either.

I think the original split was made to save on ROM and a bit of RAM too. 
However, this is done at the expense of readability and is the source of general confusion as no-one is really sure what is happening in that module when dynamic channels are enabled. Eg see https://github.com/zephyrproject-rtos/zephyr/issues/62164.

I believe we should rather have something readable and maintainable, and only then should we profile and optimize for ROM/RAM.

tims-ot, 2024-01-03:
I fully agree that readability/maintainability should take precedence over ROM/RAM optimizations.

But another thing, does having two different code paths, mean that we cannot guarantee that GATT operations will be delivered to the peer in order?
It might not matter much for a simple device sending temperature measurements, that 2 subsequent measurements are reversed.
But for a device using GATT as a data transport layer, everything can fall apart if things are not delivered in order.

jori-nordic, 2024-01-03:
Pretty much, yes. No in-order delivery guarantee, that is by design in the EATT spec.

jori-nordic, 2024-07-02:
Addressed by https://github.com/zephyrproject-rtos/zephyr/pull/72090

Buffers intended for both static and dynamic channels are now enqueued in the same code-path. The PDU headers are added just-in-time, when the lower layer (conn.c in this case) pulls HCI fragments from l2cap.

closed 2024-07-02 by jori-nordic