zephyrproject-rtos/zephyr #67145: Bluetooth: deprecate BR/EDR support

jori-nordic, 2024-01-03:
## Introduction

Deprecate Bluetooth "classic", ie BR/EDR support in the Bluetooth subsystem.

### Problem description

Why: 
- cognitive load when refactoring host and fixing bugs
- codepaths are not tested in CI -> BR is probably already broken
- with LE Audio, all the use-cases of BR/EDR are supported by Bluetooth LE
- BR/EDR security seems to get more exploits than LE
- No real maintainers / owners (company or individual)
- no reports of people using it in production

### Proposed change

Remove everything related to Bluetooth "classic" BR/EDR in the `subsys/bluetooth/host` path.
That includes dedicated C files (e.g. `rfcomm.c` etc) and conditionally compiled code in shared files (e.g. `conn.c`).

### Alternatives

- Separate the BR/EDR stack from the Bluetooth subsystem (ie no shared files, no dual-stack devices: LE + BR/EDR not supported)
- A company or individual steps up to maintain the Classic stack. That means:
  - Ability to comment on all Bluetooth host PRs (ie quick back and forth discussion)
  - Providing automated tests for all Bluetooth PRs, ideally upstream with a reference controller
    - ref controller means a given board/shield that is easy and cheap to acquire, so anyone can test locally

hongshui3000, 2024-01-03:
While I agree that it will be a matter of time before classic Bluetooth becomes obsolete, now is not the time. At least classic Bluetooth is more popular than ble for Bluetooth audio. 
I also think that the classic Bluetooth part of the zephyr Bluetooth protocol stack is not well maintained.

jori-nordic, 2024-01-03:
You are right that BR/EDR is still very much everywhere (ie 99% of audio products), and it will be multiple years until it is phased out. 
As a hobbyist myself, I'd be glad to have a working Classic Audio sink application using the zephyr stack and an espressif chip.

> no reports of people using it in production

By "it" I mean the Zephyr BR/EDR support. I haven't gotten reports of the zephyr classic stack being used in production.
My impression is that the BR part of the stack is not only not maintained, it is simply not working.

That's what I hint in the _Alternatives_ section: if a maintainer steps up and writes some tests for common use cases, and with dual-stack usage then that would also be fine.

jhedberg, 2024-01-03:
Ideally someone (or some company) should step up and maintain BR/EDR support in Zephyr. I think it might be good to send an email to the devel mailing list, to call out for support, before concluding that no-one is interested. If there really isn't anyone interested in supporting/maintaining BR/EDR then I'd be in favour of removing it (rather than splitting it into a completely independent module).

hongshui3000, 2024-01-03:
After abolishing classic Bluetooth, some interfaces for classic Bluetooth will be left.Is it possible to do some separation work (ble and classic Bluetooth), so that other people who use classic Bluetooth can continue to use their own maintained classic Bluetooth part, starting from the L2CAP layer or HCI layer.

jhedberg, 2024-01-03:
@hongshui3000 sounds like you're actively using BR/EDR? Would you be interested in becoming a maintainer for it?

There's really no way to separate BR/EDR and LE from each other except by having separate controllers for each. This is because there is overlap in the HCI commands and events that they use (e.g. HCI init sequence, HCI_Disconnect command or HCI Disconnect Complete HCI event).

hongshui3000, 2024-01-03:
> @hongshui3000 sounds like you're actively using BR/EDR? Would you be interested in becoming a maintainer for it?
>
> There's really no way to separate BR/EDR and LE from each other except by having [...]

I'm using classic bluetooth but I lack the time to maintain

In the current state, it is indeed difficult to access a private protocol stack (classic Bluetooth), and zephyr’s classic Bluetooth has not made progress for a long time.

jori-nordic, 2024-01-03:
@hongshui3000 do you also use LE in your BR/EDR apps?

If not, I think a separation might still be doable, but as johan and I said above it would not be possible to have a dual-protocol stack anymore.

hongshui3000, 2024-01-03:
> @hongshui3000 do you also use LE in your BR/EDR apps?
> 
> If not, I think a separation might still be doable, but as johan and I said above it would not be possible to have a dual-protocol stack anymore.

yes，General Bluetooth headsets(TWS), mainly classic Bluetooth, ble will be used to broadcast battery information, etc.

jori-nordic, 2024-01-03:
Alright, I stand corrected then: at least 1 report of production use :)

I will send a message to the mailing list to see if there are potential maintainers.

hongshui3000, 2024-01-03:
By the way, NXP is actually also dealing with these. After reading the code, it seems that it has also strengthened the classic Bluetooth part.
https://github.com/nxp-mcuxpresso/mcux-sdk-middleware-edgefast-bluetooth

MarkWangChinese, 2024-01-15:
Hi All, NXP has requirements for BR/EDR, and I am working on retrying to upstream the A2DP to zephyr. The previous PR are 31583 and 35536. I am preparing new PR for them, they should be ready this week.

jhedberg, 2024-01-15:
> Hi All, NXP has requirements for BR/EDR, and I am working on retrying to upstream the A2DP to zephyr. The previous PR are 31583 and 35536. I am preparing new PR for them, they should be ready this week.

It's good to hear that you're working on upstreaming new BR/EDR features like A2DP. However, what upstream needs more than anything else is somebody to step up as maintainer for BR/EDR in general, since the currently available functionality is not receiving enough attention. Would somebody from NXP be willing to step up as maintainer for BR/EDR?

jori-nordic, 2024-01-15:
@MarkWangChinese : I second @jhedberg 's comment.

It's nice to have features contributed to, but if there is no active maintainership, then those features slowly die.

Since NXP apparently depends on BR/EDR support, it would be great if someone there could be maintainer.

lylezhu2012, 2024-01-16:
Hi @jhedberg and @jori-nordic,

I come from NXP. I would like to be a maintainer of BR/EDR.

Could you give me a guidance on how to step up as maintainer for BR/EDR?

jori-nordic, 2024-01-16:
Hey @lylezhu2012 thanks!

You should read this page, in particular this paragraph: https://docs.zephyrproject.org/latest/project/project_roles.html#maintainer

You'd have to join the weekly zephyr bluetooth host meeting, where we triage bugs and discuss PRs and future changes.
-> create an account and join this: https://lists.zephyrproject.org/g/bluetooth, then you'll have access to the calendar and can join the meeting.

And I'll quote what I said above: 
> A company or individual steps up to maintain the Classic stack. That means:
- Ability to comment on all Bluetooth host PRs (ie quick back and forth discussion)
- Providing automated tests for all Bluetooth PRs, ideally upstream with a reference controller
  (ref controller means a given board/shield that is easy and cheap to acquire, so anyone can test locally)

Basically, we need tests to cover the existing BR/EDR API + the samples, and those tests should be easy to run by anyone wanting to contribute to BR/EDR.
What hardware platform are you using, is it affordable/available?

@jhedberg I don't know about the bureaucracy, does @lylezhu2012 need a minimum of accepted PRs or some other criteria before he can be added to `MAINTAINERS.yml`?

jhedberg, 2024-01-16:
> @jhedberg I don't know about the bureaucracy, does @lylezhu2012 need a minimum of accepted PRs or some other criteria before he can be added to `MAINTAINERS.yml`?

Normally there does need to be some amount of contributions, and seems like @lylezhu2012 already has several commits in the upstream tree (just one for Bluetooth though). Anyway, if we conclude that the person is a good fit (has enough basic knowledge of the subarea, understands the maintainer role responsibilities, and is committed to spending time maintaining the subarea).

@lylezhu2012 since we haven't collaborated much in the past, it would be nice to get some basic intro of yourself. As @jori-nordic mentioned, we could also do this in the weekly Bluetooth host meeting, which takes place on Thursday's.

Once we've sorted out the details, I think it's simply a matter of submitting a PR to add a new section to `MAINTAINERS.yml` for Bluetooth Classic (not sure if we want to call it that or "Bluetooth BR/EDR").

lylezhu2012, 2024-01-17:
> Hey @lylezhu2012 thanks!
>
> You should read this page, in particular this paragraph: https://docs.zephyrproject.org/latest/project/project_roles.html#maintainer
>
> You'd have to join the weekly zephyr [...]

Thanks for your sharing.

I am using RT1170-EVKB + Murata M.2 wifi-bt module. The set of hardware is available to users.

lylezhu2012, 2024-01-17:
> > @jhedberg I don't know about the bureaucracy, does @lylezhu2012 need a minimum of accepted PRs or some other criteria before he can be added to `MAINTAINERS.yml`?
> 
> Normally there does need to be some amount of contributions, and seems like @lylezhu2012 already has several commits in the upstream tree (just one for Bluetooth though). Anyway, if we conclude that the person is a good fit (has enough basic knowledge of the subarea, understands the maintainer role responsibilities, and is committed to spending time maintaining the subarea), then I don't think there should be any problem.
> 
> @lylezhu2012 since we haven't collaborated much in the past, it would be nice to get some basic intro of yourself. As @jori-nordic mentioned, we could also do this in the weekly Bluetooth host meeting, which takes place on Thursday's.
> 
> Once we've sorted out the details, I think it's simply a matter of submitting a PR to add a new section to `MAINTAINERS.yml` for Bluetooth Classic (not sure if we want to call it that or "Bluetooth BR/EDR").

Hi @jhedberg, I have joined a meeting "Zephyr: bluetooth-sig LE Audio Meeting (on LFX)". I am not sure whether it is the same one, you are mentioned. But I find it is hold every Tuesday.

jori-nordic, 2024-01-17:
> The set of hardware is available to users

But not cheap. I see it is 180$, and that's without the Bluetooth module.
Contrast this to Bluetooth LE support in zephyr, which works with 50$ nRF52840dk or even 10$ 52840 dongle. That allows even hobbyists to run the BLE tests locally and submit patches.

It would be nice if the whole reference platform (ie radio + mcu) was sub-100$. It does not need to be a full-featured or very powerful evaluation kit, just the bare minimum to run your NXP Bluetooth chipset.

What chipset specifically are you targeting, is it on one of the modules from [this page](https://www.embeddedartists.com/m2/)?

> I have joined a meeting "Zephyr: bluetooth-sig LE Audio Meeting (on LFX)".

It's the wrong one, sorry. The right one is named: "Zephyr: Bluetooth Meeting (on LFX)" and it is on Thursdays. If you don't find it, feel free to email me today (see one of my commits for email address) and I'll send you a link.

carlescufi, 2024-01-17:
@lylezhu2012 this is the meeting: https://github.com/zephyrproject-rtos/zephyr/wiki/Zephyr-Committee-and-Working-Group-Meetings#zephyr-bluetooth-meeting

lylezhu2012, 2024-01-18:
> But not cheap. I see it is 180$, and that's without the Bluetooth module. Contrast this to Bluetooth LE support in zephyr, which works with 50$ nRF52840dk or even 10$ 52840 dongle. That allows even [...]

I am using Murata [2EL M.2] (https://www.embeddedartists.com/products/2el-m-2-module/). This module can work with MCUs with UART flow control. So   RT1170-EVKB can be replaced.

> It's the wrong one, sorry. The right one is named: "Zephyr: Bluetooth Meeting (on LFX)" and it is on Thursdays. If you don't find it, feel free to email me today (see one of my commits for email [...]

> @lylezhu2012 this is the meeting: https://github.com/zephyrproject-rtos/zephyr/wiki/Zephyr-Committee-and-Working-Group-Meetings#zephyr-bluetooth-meeting

@carlescufi Thanks a lot.

jori-nordic, 2024-01-18:
@lylezhu2012 added you here: https://github.com/zephyrproject-rtos/zephyr/pull/67792

lylezhu2012, 2024-01-23:
> But not cheap. I see it is 180$, and that's without the Bluetooth module. Contrast this to Bluetooth LE support in zephyr, which works with 50$ nRF52840dk or even 10$ 52840 dongle. That allows even [...]

@jori-nordic,

I planed to use the boards [NXP LPCXPRESSO55S69](https://www.nxp.com/design/design-center/software/development-software/mcuxpresso-software-and-tools-/lpcxpresso-boards/lpcxpresso55s69-development-board:LPC55S69-EVK) ($44.85)  + [2EL M.2](https://www.embeddedartists.com/products/2el-m-2-module/) ($36) + [uSD-M2 Adapter LBEE0ZZ1WE-uSD-M2](https://www.murata.com/en-global/products/connectivitymodule/wi-fi-bluetooth/overview/lineup/usd-m2-adapter) ($68.68) to support Bluetooth Classic. 

But I find the total price is $149.53. It is more than $100. 

I would like to ask, does the total price have to be within $100?

Do you have any suggestions for a reference platform?

lylezhu2012, 2024-01-23:
> @jori-nordic,
>
> I planed to use the boards [NXP [...]

Another option is, [MIMXRT1040-EVK](https://www.nxp.com/design/design-center/development-boards/i-mx-evaluation-and-development-boards/i-mx-rt1040-evaluation-kit:MIMXRT1040-EVK) ($90.85) + [2EL M.2](https://www.embeddedartists.com/products/2el-m-2-module/) ($36). The total price is $126.85. This option is more suitable as it does not require any Dupont wire connections.

jhedberg, 2024-01-23:
@jori-nordic I guess we can close this one now as "not planned"?

jori-nordic, 2024-01-24:
Closing as "Not planned" (anymore). We have a new BR/EDR maintainer.

jori-nordic, 2024-05-07:
> Another option is, MIMXRT1040-EVK ($90.85) + 2EL M.2 ($36). The total price is $126.85. This option is more suitable as it does not require any Dupont wire connections

@lylezhu2012 Nordic provided me with two of these boards for testing my patches. I cannot find them in the zephyr tree though. Are you using a fork of Zephyr with support for them?

closed 2024-05-07 by alwa-nordic

lylezhu2012, 2024-06-03:
> @lylezhu2012 Nordic provided me with two of these boards for testing my patches. I cannot find them in the zephyr tree though. Are you using a fork of Zephyr with support for them?

I created a [PR](https://github.com/zephyrproject-rtos/zephyr/pull/70815) to support RT1170 EVKB + 2EL.