zephyrproject-rtos/zephyr #67169: boards: doc: Deadlink in SECO JUNO SBC-D23 documentation page ('boards/arm/stm32f3_seco_d23/doc/index')

kartben, 2024-01-03:
**Describe the bug**

SECO JUNO SBC-D23 [documentation page](https://docs.zephyrproject.org/latest/boards/arm/stm32f3_seco_d23/doc/index.html) contains a deadlink to https://edge.seco.com/juno.html

Could it be that this board now goes by a new name, @ektor5? Looking at the board image, it feels like it's now https://edge.seco.com/fr/sbc-3-5-px30.html? 

**Environment**

* Page: `boards/arm/stm32f3_seco_d23/doc/index`
* Version: main
* SHA-1: 14c92da96d5613b48610fa9bfe284528a814a4e6

ektor5, 2024-01-04:
Hi @kartben ,
Affermative, I'm surprised there are no redirects from old links.
We have no choice but replace old link with the new one https://edge.seco.com/sbc-3-5-px30.html.
Thanks for the heads up,
Ettore

kartben, 2024-01-04:
Thanks for the quick feedback! Would you be able to please send a PR when you have a chance? Would it make sense to also rename the board to "SBC-3.5-PX30" then?

ektor5, 2024-01-04:
Anyway, with the name change, a doc review is necessary, I'll open a MR.