zephyrproject-rtos/zephyr #67171: net: ppp: doc: PPP documentation page ('connectivity/networking/api/ppp') needs updating

kartben, 2024-01-03:
**Describe the bug**

Spotted two issues with [PPP documentation page](https://docs.zephyrproject.org/latest/connectivity/networking/api/ppp.html) while looking for dead links in the documentation:

- "See also the [samples/net/sockets/echo_server/overlay-ppp.conf](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/sockets/echo_server/overlay-ppp.conf) file for configuration option examples." is a dead link.
- "The PPP support in Zephyr 2.0 is still experimental [...]" sentence needs updating

**Environment**

* Page: `connectivity/networking/api/ppp`
* Version: main
* SHA-1: 69d19ae9fb68d0be38ceb22ba43c9fff985cbbaa

cfriedt, 2024-01-09:
@jukkar - a really long time ago, I had a development branch that moved HDLC to its own library, since it can be used even without the IP layer. I know of one product that used that over a board-level uart connection, but that's mainly it.

The nice thing about a library implementation is that it can be separately tested. Would that be a worthwhile change?

jukkar, 2024-01-10:
> The nice thing about a library implementation is that it can be separately tested. Would that be a worthwhile change?

Sure, that sounds like a good change, please submit to upstream for a review.

PR #67252 (2024-01-10): Update the ppp documentation
Fixes #67171
commit b0dd9a29 (2024-01-10): doc: net: ppp: Fix the point-to-point documentation

The ppp doc was still referring to gsm-modem sample which
is no longer there. Also removed old information that is no
longer relevant.

Fixes #67171
closed 2024-01-10 by carlescufi