zephyrproject-rtos/zephyr #67178: tests: posix: pthread_attr_destroy() not consistently called after pthread_attr_init()

cfriedt, 2024-01-03:
**Describe the bug**

I have not looked into how other POSIX implementations manage thread stacks (as per the standard POSIX threading model), but it is becoming somewhat clear that it should be done as part of `pthread_attr_init()` and `pthread_attr_destroy()`.

Dynamic thread stacks (including pooled thread stacks) were not done in Zephyr this way due to... difficulties getting that accepted in the first place (it took years), but also due to the implementation of the POSIX API, testsuite, and so on.

The model is quite clear though - either use a NULL `pthread_attr_t *` and the implementation will manage the stack for the caller, or call `pthread_attr_init()` **with a matching call** to `pthread_attr_destroy()`.

In many areas in the POSIX testsuite, calls to `pthread_attr_destroy()` were missing or an assertion would occur prior to `pthread_attr_destroy()` being called.

With recent work in Zephyr, it has become quite obvious that the testsuite leaks memory and resources substantially due to this.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? All
 - What have you tried to diagnose or workaround this issue? #67094 #67085
 - Is this a regression? Yes? No? It's hard to say. The POSIX API has been in development for years. Is a regression a regression if it has not been caught? Then yes, otherwise no.

**To Reproduce**
Steps to reproduce the behavior:
1. Pull #67085, and in particular, the commit which moves `struct posix_thread_attr` into `struct posix_thread`
2. execute tests where there is a call to `pthread_attr_init()` but no matching call to `pthread_attr_destroy()`, or tests that assert before calling `pthread_attr_destroy()` (basically all of them)
3. See error

**Expected behavior**
There should be no memory leaks

**Impact**
It's a showstopper for #51211 which is why I'm in such a hurry to fix it 😁 

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK v0.16.4
 - Commit: c0bb064d4e

**Additional context**

cfriedt, 2024-01-03:
Intentionally did not select high priority here, even though it affects all platforms, since it's being actively worked on

cfriedt, 2024-01-03:
> Intentionally did not select high priority here, even though it affects all platforms, since it's being actively worked on

Actually no, I'll mark this as high. Maybe people will stop accusing me of trying to fast-track things if I escalate priority when appropriate. It affects multiple platforms.

cfriedt, 2024-01-04:
Not completely fixed yet...

closed 2024-01-26 by cfriedt

jobol, 2025-01-23:
If I understand correctly, the pthread_attr_t instance must live all long the thread extend.

So the following code is bad:

```C
int x_thread_create(
			x_thread_t *tid,
                        x_thread_cb entry,
			void *arg,
			int detached)
{
        int rc;
	pthread_attr_t attr;

	pthread_attr_init(&attr);
	if (detached)
		pthread_attr_setdetachstate(&attr, PTHREAD_CREATE_DETACHED);
	rc = pthread_create(tid, &attr, (void*(*)(void*))entry, arg) ? -errno : 0;
	pthread_attr_destroy(&attr);
       return rc;
}
```

I reached it because logs are telling `<err> os: tid 0x20001418 is in use!`
And IMHO it leads to memory leak because when the thread stops, k_thread_stack_free isn't called.

Can you confirm my analysis please?

cfriedt, 2025-01-23:
@jobol - I'm not sure what the analysis is here or what the goal is for that matter.

This is also not really a great place to start a random discussion.

Have you considered asking your question(s) on Discord?

jobol, 2025-01-23:
@cfriedt - hi thak you for your quick reply. I was searching for a bug related to what I was facing. This bug is very related to that issue and I wanted to have your feedback on that topic. No I didn't consider asking on discord because I'm not on discort but I can do it as well. Nevetheless you are probably the most involved man on that topic and your feedback on my question would be very appreciated.