zephyrproject-rtos/zephyr #67181: [RFC] posix: add kconfig and cmake for specifying application conformance

cfriedt, 2024-01-03:
## Introduction

This RFC aims to establish a requirement for the way that Zephyr applications and libraries use reserved, standard POSIX application conformance macros. Such macros are typically defined _by the application_ rather than the implementation and include macros such as:

* `_POSIX_C_SOURCE`
* `_XOPEN_SOURCE`

Applications and libraries written to take advantage of POSIX APIs are written according to a version or range of versions of the POSIX API.

On multi-purpose / desktop POSIX systems that are capable of self-hosting, this configuration is usually baked into the toolchain or environment so the user does not explicitly need to specify which POSIX API version (or range of versions) the application conforms to.

There is a mechanism by which the application advertises it's POSIX conformance level, namely by defining `_POSIX_C_SOURCE=200809L` or `_XOPEN_SOURCE=700`. In this case, those two constants indicate that the application is written in conformance with IEEE 1003.1-2017. Earlier versions would be e.g. `200112L` / `600` for IEEE 1003.1-2001, etc.

The implementation uses these macros in order to expose the appropriate `#define`s, structures, and prototypes to the application via C header files, which overlap with ISO C header files. If the ISO C header files do not make the necessary POSIX declarations, then `#include_next` is required.

In self-hosting systems (i.e. those which are also capable of providing a POSIX shell, compiler, and utilities), most of this is baked into the toolchain specs. The headers remain the same (prototypes are guarded by `_POSIX_C_SOURCE` or derivatives thereof) and are switched on by built-in compiler flags.

Indeed, the POSIX specification itself says that application conformance macros (as well as implementation conformance macros) should be defined prior to any header being included. There are a few ways to do that.

1. Toolchain specs (requires rebuilding the toolchain, does not take user-configuration into account, so is not super practical)
2. Via compiler flags such as `-D_POSIX_C_SOURCE=200809L` or similar (a friendly cmake file might check to see if this is defined before redefining it)
3. Using compiler flags such as the `-include foo.h` directive (although this is possibly cheating)
4. Including a private, local header file prior to including (although this is possibly cheating)
5. Via source-level redefinitions
```cpp
#if !defined(_POSIX_C_SOURCE) || _POSIX_C_SOURCE < 200112L
#undef _POSIX_C_SOURCE
#endif
#define _POSIX_C_SOURCE 200112L
```

### Problem description

One might wonder, why is this even an RFC? Shouldn't this just be part of the POSIX API? That's a very valid question.

The major problem(s) arise when other libraries that are part of the Zephyr build use something like the following:

```cmake
zephyr_compile_definitions(_POSIX_C_SOURCE=200809)
zephyr_library_compile_definitions(_POSIX_C_SOURCE=200112L)
```

To expose POSIX API prototypes.

There are 2 main reasons why this is a problem.

1. `200809` is not the value required by IEEE 1003.1-2017 (it's actually `200809L`), so it is implicitly conflicting with other attempts to globally define the standard value
2. `zephyr_compile_definitions()` adjusts the **global** compiler flags, which prevents **the application** from being able to select it's conformance level in a scalable way
3. `zephyr_library_compile_definitions()` adjusts compiler flags, which prevents **the application** from being able to select it's conformance level in a scalable way

The first item is an issue with `CONFIG_COMPILER_WARNINGS_AS_ERRORS=y` since it will generate a redefinition warning that is escalated to error at build time if an application were to declare the _correct_ version of that reserved and standard application conformance macro.

Between the lines, we want Zephyr application developers to be able to quickly and conveniently select the level of POSIX application conformance (and yes, not defining any macro is a legitimate option). However, when we are looking at 10000 different tests, sample applications, downstream apps, etc, we actually _do_ care about 10000 x 4 = 40000 lines of code that we would be pushing onto _Zephyr users_ just to effectively have a workaround.

This leaves us with few options other than a global definition.

As a result, the second and third items are problems again, because we do not want to have redefinition errors on global macros (those that are declared on the command line), since warnings can escalate to errors.

> [!NOTE]  
> Exactly zero % of this RFC has imposed anything on how Zephyr's POSIX API features are linked into the application.
> That means, that choosing the application conformance level is independent of using Kconfig and CMake to link sources
> into the final image. This allows implementations of features to come from the POSIX API, from the C library, from user code
> or anywhere else, which is consistent with the way that C programs are written in any other environment.

> [!NOTE]  
> Contrary to application conformance macros, _implementation_ conformance macros (aka feature test macros)
> are reserved to be defined exclusively by the implementation. Such macros include 

> [!NOTE]  
> This is quite critical for the [LTSv3 POSIX Roadmap](https://github.com/zephyrproject-rtos/zephyr/issues/51211), which is otherwise quite on track and has been planned far longer than other conflicting areas have been conflicting.
[... 42 lines cut ...]
AFAIK, there are no other libraries in the Zephyr repository itself that have potentially conflicting definitions 

Other parts of Zephyr, most importantly the application / user, should have no idea that any aspect of the build has changed, aside from maybe 

### Concerns and Unresolved Questions

- [ ] Are there other Zephyr modules that define standard, reserved POSIX application conformance macros in conflicting ways? If so, what are they?

## Alternatives

The status quo, which has.. unpleasant results

## References

### Standards

* [IEEE 1003.1-2017](https://standards.ieee.org/ieee/1003.1/7101/)
* [OpenGroup.org](https://pubs.opengroup.org/onlinepubs/9699919799/)

### Issues

* https://github.com/zephyrproject-rtos/zephyr/issues/51211
* https://github.com/zephyrproject-rtos/zephyr/issues/66921
* https://github.com/zephyrproject-rtos/zephyr/issues/67041
* https://github.com/picolibc/picolibc/issues/649

keith-packard, 2024-01-03:
This doesn't make it clear whether these new compile definitions are global, per-directory or per-file. 

Many C libraries support multiple APIs simultaneously, so you can build one file with the POSIX api and another with the GNU api and they will link and run correctly.  Picolibc, in particular, provides native Zephyr API support while also providing POSIX API support when applications set `_POSIX_C_SOURCE`.

As the API definitions affect how individual files are compiled, I'd strongly recommend limiting how many files are affected by the defines. In particular, the core Zephyr code should *never* expect to be built against the POSIX API, only the Zephyr API.

I think part of the confusion here is just how POSIX APIs are intended to be supported in Zephyr. Some of them are OS-level abstractions (read/write/open/close/lseek) which the C library cannot implement. In the middle are things like ndbm which provide significant functionality but which cannot operate without an underlying OS. Then there are portions of the API which are entirely independent of the OS (e.g. regex) and can be provided by the C library with no need for any underlying OS interface.

Is the goal to implement *all* of the POSIX API which is not in the C standard within the Zephyr code base? Or is the goal to implement only the OS-level abstractions which a C library cannot implement without deep OS knowledge? Frankly, I'd argue for the latter -- re-implementing parts of the POSIX spec which already exist in numerous C libraries which Zephyr supports seems like a poor use of resources, as well as generating a large support burden for code which probably won't get a lot of use (do you know of any applications which use ndbm?).

Once you've decided where various POSIX apis should be implemented, the remaining task is how to get the header files configured so that POSIX code can see those APIs simply by defining `_POSIX_C_SOURCE` before including any of them. The problem here is that POSIX adds many of its API definitions to header files that are already defined in the C standard. So a C library used within Zephyr which offers only the C standard APIs may not define POSIX APIs when `_POSIX_C_SOURCE` is set. I see two alternatives here:

 1. Shadow the C library headers within Zephyr, using  `#include_next` within them to chain to the C library version.
 2. Push POSIX API definitions upstream to the C library sources so that the native headers work out of the box.

It seems like the second solution would be ideal; with no hacks required within the Zephyr tree, and so we should prefer that option where possible. For C libraries without Zephyr support, creating shadow headers within the Zephyr tree that make them "zephyr-compatible" may be the best solution. These  C-library specific shadow headers could also ensure that the Zephyr API was visible when compiling native Zephyr code. A fairly generic "ISO C library wrapper" set would probably work for most pure C libraries from random vendor SDKs.

cfriedt, 2024-01-04:
> This doesn't make it clear whether these new compile definitions are global, per-directory or per-file.

The compile definitions could go into a global Cmake variable, as mentioned.

Those global Cmake variables can then be used, not used, or derived from, for building local targets (whatever CMake supports - so per file, directory, anything) as long as the local build flags do not override the CMake variable or conflict with other settings.

I hope that clarifies your question.

But again, the application can be built with any defines that the application wants, so if the application (including downstream variants) wants to globally define _POSIX_C_SOURCE, there is little that anyone can do about it. We should really do our best to ensure that the build is not broken due to conflicting application-centric definitions.

> Many C libraries support multiple APIs simultaneously, so you can build one file with the POSIX api and another with the GNU api and they will link and run correctly.

I don't see what is proposed as being a limiting factor whatsoever in that scenario.

> Picolibc, in particular, provides native Zephyr API support while also providing POSIX API support when applications set `_POSIX_C_SOURCE`.

Yes.

> As the API definitions affect how individual files are compiled, I'd strongly recommend limiting how many files are affected by the defines. In particular, the core Zephyr code should *never* expect to be built against the POSIX API, only the Zephyr API.

That should fall completely within the scope of what is described here.

> I think part of the confusion here is just how POSIX APIs are intended to be supported in Zephyr.

Again, zero % of this RFC imposes anything about what sources are being built. So I'm going to consider this question as off topic.

> things like ndbm which provide significant functionality but which cannot operate without an underlying OS.

This is also off-topic.

> Then there are portions of the API which are entirely independent of the OS (e.g. regex) and can be provided by the C library with no need for any underlying OS interface.

Again, zero % of this RFC imposes anything about what sources are being built. So I'm going to consider this question as off topic. 

> Is the goal to implement *all* of the POSIX API which is not in the C standard within the Zephyr code base?

Off topic, but no. Just what is required for PSE51, 52, and parts of 53.

> Or is the goal to implement only the OS-level abstractions which a C library cannot implement without deep OS knowledge?

The goal is to provide a consistent POSIX API to augment what is available in the C library to the extent that is required by PSE51, PSE53, and part of PSE53.

Of course, not every C library implements POSIX, since POSIX is not a part of ISO C, and POSIX clumsily re-used standard headers instead of being orthogonal. `#include_next` to the rescue!

> Once you've decided where various POSIX apis should be implemented

This is off topic..

I did add a reference to #51211 but I will also add a reference to
https://docs.zephyrproject.org/latest/services/portability/posix/index.html

> the remaining task is how to get the header files configured so that POSIX code can see those APIs simply by defining `_POSIX_C_SOURCE` before including any of them.

That's what is outlined in this RFC.

> The problem here is that POSIX adds many of its API definitions to header files that are already defined in the C standard.

Yep.

I think `#include_next` can be used for that purpose. It was in this RFC originally, but I removed it because I didn't want people to be too confused.

> So a C library used within Zephyr which offers only the C standard APIs may not define POSIX APIs when `_POSIX_C_SOURCE` is set. I see two alternatives here:

>  1. Shadow the C library headers within Zephyr, using  `#include_next` within them to chain to the C library version.

Yes, that is one solution.

>  2. Push POSIX API definitions upstream to the C library sources so that the native headers work out of the box.

Sure... Except that won't work for some "upstream" C libraries because they might be proprietary.

> It seems like the second solution would be ideal;

Ideal, yes.

> with no hacks required within the Zephyr tree, and so we should prefer that option where possible.

Agreed

> For C libraries without Zephyr support, creating shadow headers within the Zephyr tree that make them "zephyr-compatible" may be the best solution.

Agreed (the rest too)

aescolar, 2024-01-08:
Some extra background points for the discussion:

* It is also possible to undefine these with compiler command line arguments/build options.
* Defining these flags globally exposes the whole API globally, which also prevents checkpatch from detecting breaches of rules A.4 and A.5.

--------

My take is that setting these macros globally from the Zephyr tree should be avoided.
(I'd fix our libraries/source so we don't do it, don't need it, and they are resiliant to the app setting them to something else).

So my proposal would be to:
a) Make Zephyr source, for its own use, as independent as possible from what anything else may have set: Libraries/individual C files which need any _SOURCE macro, undefine and define what they need either thru build/cmake or at the C source.
b) These Zephyr libraries should not set these macros globally but only in their scope as they need.

Wrt. to the expectation of Zephyr having more than the plain C std library visible by default, I would think that should be corrected to not be the case. But that is something the C library maintainers would need to agree and decide. So, also:
c) C libraries should not set these globally as a workaround.

I would not be opposed to defining a cmake property to carry this option for the application, though the benefit of it would be much smaller if we do a) and b) (and also c).
Even if we define that property, I still think the Zephyr tree source should do a) and b), and not use that property. Using that property still relies on nobody setting the flag globally in another way.

-----

Sidenote,
This first line
```
#if !defined(_POSIX_C_SOURCE) || _POSIX_C_SOURCE < 200112L //<--
#undef _POSIX_C_SOURCE
#endif //<--
#define _POSIX_C_SOURCE 200112L
```
is only needed if this is in a header the application/another library may use. Otherwise just undefining and defining is enough.

-----

> I think part of the confusion here is just how POSIX APIs are intended to be supported in Zephyr. Some of them are OS-level abstractions .. which the C library cannot implement .. Then there are portions of the API which are entirely independent of the OS (e.g. regex)..

I very much agree with @keith-packard that we should not conflate these 3 by default, but make a concious decision about which are provided from where. Relaying solely on the C library for the later type and improving that library when something is missing seems like a good idea.
Otherwise a solution where OS agnostic parts are kept in separate libraries with their own kconfig+cmake selection, with a default selection for the Zephyr provided C libraries, would seem reasonable.

cfriedt, 2024-01-09:
Sorry for taking so long in the arch meeting. Thanks @aescolar for attending as well. I think in general what was discussed shouldn't pose any conflicts.

We definitely want to avoid adding global definitions, but also want to have a simple and concise way for users to specify local definitions in a way that scales well.

I'll put together a PR with the concept cmake extension + Kconfig that I mentioned.

I think this can be done in a way that is consistent with everyone's ideas and does not pose any conflicts.

What Alberto also mentioned about maybe refining CI checks seems like a good idea to me too, and was also mentioned in #67283 .

cfriedt, 2024-02-01:
PR #67132 has been updated. I added some screenshots and rebased on main. Made the menuconfig items a bit prettier, but that's about it.

It should probably go in before v3.6.0.