zephyrproject-rtos/zephyr #67184: STM32WBA LPUART1/USART2 GPDMA receive error

MY201314MY, 2024-01-04:
**Describe the bug**
We obtained the nucleo_wba52cg development board from the ST official website. This board can be officially supported by zephyr.
https://www.st.com/en/evaluation-tools/nucleo-wba52cg.html
https://docs.zephyrproject.org/latest/boards/arm/nucleo_wba52cg/doc/nucleo_wba52cg.html
According to this submission record
https://github.com/zephyrproject-rtos/zephyr/pull/63816/files
We believe that zephyr provides support for GPDMA. At present, there is no problem with serial port sending.But as for receiving except for the first time it has always been wrong afterwards.
[image]
In this example I sent AT/def/OPQ/RST
But only the first AT is received correctly. In other receptions, it can recognize the correct number of characters.
**To Reproduce**
Steps to reproduce the behavior:
*zephyr version*
```shell
pi@inspiron:~$ cd zephyr/zephyr/
pi@inspiron:~/zephyr/zephyr$ git branch
* main
pi@inspiron:~/zephyr/zephyr$ git log -n 1
commit 89982b711bbda3cdeb2da6a7250ccab4f088ba10 (HEAD -> main, origin/main, origin/HEAD)
Author: TOKITA Hiroshi <tokita.hiroshi@fujitsu.com>
Date:   Sat Dec 30 21:35:43 2023 +0900

    drivers: serial: ra: minor cleanups
    
    - Corrected indentation with clang-format
    
    Signed-off-by: TOKITA Hiroshi <tokita.hiroshi@fujitsu.com>
pi@inspiron:~/zephyr/zephyr$ 
```

**nucleo_wba52cg.overlay**
```c
/ {
	chosen {
		zephyr,console = &usart1;
		zephyr,shell-uart = &usart1;
	};
};

&usart2 {
	pinctrl-0 = < &usart2_tx_pa12 &usart2_rx_pa11 >;
	pinctrl-names = "default";
	status = "okay";

	dmas = <&gpdma1 0 14 STM32_DMA_PERIPH_TX
		&gpdma1 1 13 STM32_DMA_PERIPH_RX>;
	dma-names = "tx", "rx";
};

&lpuart1 {
	pinctrl-0 = < &lpuart1_tx_pa2 &lpuart1_rx_pa1 >;
	pinctrl-names = "default";
	status = "okay";

	dmas = <&gpdma1 0 16 STM32_DMA_PERIPH_TX
		&gpdma1 1 15 STM32_DMA_PERIPH_RX>;
	dma-names = "tx", "rx";
};

&gpdma1 {
	status = "okay";
};
```
**main.c**
```c
/*
 * Copyright (c) 2020 Nordic Semiconductor ASA
 *
 * SPDX-License-Identifier: LicenseRef-Nordic-5-Clause
 */

#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/logging/log.h>
#include <zephyr/drivers/uart.h>

LOG_MODULE_REGISTER(main, LOG_LEVEL_DBG);

#define BUF_SIZE 64
static K_MEM_SLAB_DEFINE(uart_slab, BUF_SIZE, 3, 4);

static void uart_callback(const struct device *dev,
			  struct uart_event *evt,
			  void *user_data)
{
	struct device *uart = user_data;
	int err;

	switch (evt->type) {
	case UART_TX_DONE:
		LOG_INF("Tx sent %d bytes", evt->data.tx.len);
		break;

	case UART_TX_ABORTED:
		LOG_ERR("Tx aborted");
		break;

	case UART_RX_RDY:
		LOG_INF("Received data %d bytes", evt->data.rx.len);
		LOG_HEXDUMP_DBG(evt->data.rx.buf, evt->data.rx.len, "recv");

		break;

	case UART_RX_BUF_REQUEST:
	{
		uint8_t *buf;

		err = k_mem_slab_alloc(&uart_slab, (void **)&buf, K_NO_WAIT);
		__ASSERT(err == 0, "Failed to allocate slab");

		err = uart_rx_buf_rsp(uart, buf, BUF_SIZE);
		__ASSERT(err == 0, "Failed to provide new buffer");
		break;
	}

	case UART_RX_BUF_RELEASED:
		k_mem_slab_free(&uart_slab, (void *)evt->data.rx_buf.buf);
		break;

	case UART_RX_DISABLED:
		break;

	case UART_RX_STOPPED:
		break;
	}
}

static void async(const struct device *lpuart)
{
	uint16_t i = 0;
	uint8_t txbuf[128] = "I'm from nucleo_wba52cg.\r\n";
	int err;
	uint8_t *buf;

	err = k_mem_slab_alloc(&uart_slab, (void **)&buf, K_NO_WAIT);
	LOG_DBG("L:%d -- error:%d\r\n", __LINE__, err);

	err = uart_callback_set(lpuart, uart_callback, (void *)lpuart);
	LOG_DBG("L:%d -- error:%d\r\n", __LINE__, err);

	err = uart_rx_enable(lpuart, buf, BUF_SIZE, 10000);
	LOG_DBG("L:%d -- error:%d\r\n", __LINE__, err);

	while (1) {
		//sprintf(txbuf, "I'm from nucleo_wba52cg.:%d\r\n", i++);
		//err = uart_tx(lpuart, txbuf, strlen(txbuf), 10000);
		//LOG_DBG("L:%d -- error:%d\r\n", __LINE__, err);

		k_sleep(K_MSEC(5*1000));
	}
}

int main(void)
{
	const struct device *uart = DEVICE_DT_GET(DT_NODELABEL(lpuart1));

	LOG_DBG("%p", uart);

	async(uart);	

	return 0;
}
```
***prj.conf***
```shell
CONFIG_GPIO=y
CONFIG_LOG=y
CONFIG_SERIAL=y
CONFIG_DMA=y
#CONFIG_SHELL=y

CONFIG_UART_ASYNC_API=y

CONFIG_DMA_LOG_LEVEL_DBG=y
CONFIG_UART_LOG_LEVEL_DBG=y
CONFIG_UART_CONSOLE_LOG_LEVEL_DBG=y
```
**build**
```shell
west build -b nucleo_wba52cg -p
```
**Expected behavior**
The DMA UART receive can work normally.

**Additional context**
If you open CONFIG_SERIAL like this in prj.conf the debug will output nothing which is so strange.
```shell
CONFIG_GPIO=y
CONFIG_LOG=y
CONFIG_SERIAL=y
CONFIG_DMA=y
CONFIG_SHELL=y

CONFIG_UART_ASYNC_API=y

CONFIG_DMA_LOG_LEVEL_DBG=y
CONFIG_UART_LOG_LEVEL_DBG=y
CONFIG_UART_CONSOLE_LOG_LEVEL_DBG=y
```

ajarmouni-st, 2024-01-18:
@MY201314MY Hi, it seems that console problem is related to CONFIG_UART_ASYNC_API in prj.conf, which affects Shell Backend config (line 50 in /local/home/jarmouna/zephyrproject/zephyr/subsys/shell/backends/Kconfig.backends).

For now, I don't see a way to make the CONFIG_UART_ASYNC_API flag apply to a specific UART (LPUART in your case), & also pnly apply to Shell if it is associated with that UART.

ajarmouni-st, 2024-01-18:
@MY201314MY Hi, it seems that the console problem is related to CONFIG_UART_ASYNC_API in prj.conf, which affects Shell Backend config (line 50 in subsys/shell/backends/Kconfig.backends).

For now, I don't see a way to make the CONFIG_UART_ASYNC_API flag apply to a specific UART (LPUART in your case), & also pnly apply to Shell if it is associated with that UART.

ajarmouni-st, 2024-02-02:
> But the problem Nothing log output when enable **CONFIG_SHELL=y** still exists.

Hi @MY201314MY , the following patch fixes the shell issue.

```
--- a/subsys/shell/backends/Kconfig.backends
+++ b/subsys/shell/backends/Kconfig.backends
@@ -47,9 +47,16 @@ config SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN
        default y
        depends on SERIAL_SUPPORT_INTERRUPT
 
+config SHELL_ASYNC_API
+       bool "Asynchronous shell API"
+       default n
+       depends on UART_ASYNC_API
+       help
+         This option enables asynchronous shell API.
+
 choice SHELL_BACKEND_SERIAL_API
        prompt "Mode"
-       default SHELL_BACKEND_SERIAL_API_ASYNC if UART_ASYNC_API
+       default SHELL_BACKEND_SERIAL_API_ASYNC if SHELL_ASYNC_API
        default SHELL_BACKEND_SERIAL_API_INTERRUPT_DRIVEN if SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN
        default SHELL_BACKEND_SERIAL_API_POLLING
```

When `CONFIG_SHELL_ASYNC_API=y`, the user should simply assign `zephyr,shell-uart` in board DTS a different UART than the one used with `zephyr,console`.

closed 2024-02-06 by henrikbrixandersen