zephyrproject-rtos/zephyr #67187: doc: Documentation issue in 'index'

spectrum70, 2024-01-04:
**Describe the bug**

On Brave 1.61.101 there is no dropdown on the left side to select the version. As stated in the introduction, there should be one.

**Environment**

* Page: `index`
* Version: main
* SHA-1: e908f889208531ef2a64163380bda1a3a8b33d23

nordicjm, 2024-01-04:
Seems unlikely to be a problem with the html since this is a basic feature and no one else including people using other chromium-based browsers has reported such an issue. Are you sure you don't have javascript disabled or some plugin that is removing elements?

kartben, 2024-01-04:
Could it be that the mention of a "drop-down" menu is confusing? You might not be looking in the right place, @spectrum70?

[image]

kartben, 2024-01-04:
Further on the confusing wording: on mobile/narrow screens, the "dropdown" is to be found in the collapsible menu, not strictly speaking in the "menu on the left".

kartben, 2024-01-04:
I went ahead and already adjusted the title of this issue. @spectrum70 please confirm it is "just" a formulation issue and not an actual issue you might be experiencing with your particular browser / setup. Thanks!

closed 2025-06-22 by kartben