zephyrproject-rtos/zephyr #67214: Zephyr bot: removing self from PR reviewer list doesn't work

jori-nordic, 2024-01-04:
**Describe the bug**

When a github username is listed under `collaborators` for a given path in the `MAINTAINERS.yml` file, the Zephyr bot will add that github username to the list of Reviewers for a PR touching this path.

If the user removes itself from the reviewer list, the Zephyr bot will add them back to the list on the next push to the PR branch.
 
**To Reproduce**

1. Open a PR on a file a third-party is collaborator on
2. Have that user remove itself from the reviewers
3. Edit the file and force-push to the PR branch
4. See that the user is re-added by Zephyr bot.

**Expected behavior**

Zephyr bot should respect the decision and not add the manually-unsubscribed user back.

**Impact**

Annoyance, notification fatigue.
Basically makes the "Review requests" github pane unusable, need to come up with curated filter to avoid it.

henrikbrixandersen, 2024-01-04:
This is intended. One of the responsibilities as a collaborator (according to https://docs.zephyrproject.org/latest/project/project_roles.html#collaborator) is "Responsibility to review relevant code changes within reasonable time."

carlescufi, 2024-01-05:
> This is intended. One of the responsibilities as a collaborator (according to https://docs.zephyrproject.org/latest/project/project_roles.html#collaborator) is "Responsibility to review relevant code changes within reasonable time."

I agree, but we do not currently have enough resolution in the maintainers file to capture the nuances of collaborators and their areas of expertise. Not all collaborators of a huge subsystem like Bluetooth know all the sub-areas of it. So, IMHO, it only makes sense that collaborators (but not maintainers!) may opt out of reviewing certain PRs.

henrikbrixandersen, 2024-01-05:
> Not all collaborators of a huge subsystem like Bluetooth know all the sub-areas of it.

Bluetooth is already divided in the following sub-areas in the maintainers file:
- Controller
- Host
- Mesh
- Audio

Further division can be done if needed. CC: @jhedberg

carlescufi, 2024-01-05:
> Bluetooth is already divided in the following sub-areas in the maintainers file:

Correct but notice that there are many collaborators in each area, and some of the areas (Host for example) are massive. Further subdivision is not necessary nor trivial to do, because the layers are somewhat intertwined and also in many cases would partially overlap if we broke them up. I really think the best solution is to just allow collaborators to opt out of certain reviews.

henrikbrixandersen, 2024-01-05:
> Correct but notice that there are many collaborators in each area, and some of the areas (Host for example) are massive. Further subdivision is not necessary nor trivial to do, because the layers [...]

In that case, we may need to update our project role descriptions.

carlescufi, 2024-01-05:
> In that case, we may need to update our project role descriptions.

Right, I am happy to discuss that together with @jhedberg in the Process WG.

henrikbrixandersen, 2024-01-10:
> someone else from my team already reviewed / is reviewing it. distributing PRs to review over the team saves time

GitHub supports teams being assigned as reviewers. Perhaps we can start utilizing GitHub teams more?

jori-nordic, 2024-01-10:
The issue I have is this. Arguably it's a UX issue in github, but the zephyr bot exacerbates it:
I will not interact with half of the PRs in this view:

[image]

keith-zephyr, 2024-01-10:
Process WG AIs
@carlescufi to investigate tracking when users self-remove from a PR and look at having Zephyrbot honor this request
@keith-zephyr - provide examples of PRs assigning reviewers that are not a collaborator/maintainer

-----
@henrikbrixandersen - why are collaborators not reviewing PRs.  Why would they need to remove themself from a PR?
@carlescufi - in practice collaborator areas may be too broad.  Collaborators may only know about a subset of an area.
@fabiobaltieri - Jori wants to opt-out of PRs that he isn't interested in to have a cleaner dashboard.
@nashif - removing reviewers could be introduce problems
@henrikbrixandersen - another request to add notifications when an PR hits a specific area.  If we have collaborators that are present only for notifications causes problems.
@fabiobaltieri - can we improve the granularity of the Zephyrbot?
@keith-zephyr I have seen reviews assigned to me for areas that I'm not a collaborator.
@henrikbrixandersen - this could be coming from the "suggested reviewers" dialog when the PR is created.
@keith-zephyr - to look for examples.
@carlescufi - Jonathan's issue is probably more about GitHub notifications limitatons.
@fabiobaltieri - most users have developed/used their own tools to manage GitHub notifications
@nashif - uses the GitHub app to get phone push notifications - it's more targeted than the emails.
@dleach02 - phone notifications not a good flow for him
@dleach02 - too many email notifications led to routing to separate folder but it's easy to ignore the separate folder
@nashif - this is a possible enhancement we could make to Zephyrbot.
@fabiobaltieri - need to confirm that there is a specific GitHub event for removing a reviewer
@dleach02 - can the bot log the reason why someone was added as a reviewer (specific file for instance).
@keith-zephyr - can we ask Jonathon to suggest a change to the Zephyrbot to exclude users that remove themselves.
@carlescufi - would this be accepted by Zephyr?  @henrikbrixandersen - can GitHub provide the right signal.
@carlescufi - to investigate if this is possible.
@nashif - we also need improvements to fix west/mcuboot reviews.  There is some work needed to correlate the module with the correct area.
@henrikbrixandersen there are labels assigned based on the module.  @nashif the manifest action has this info.
@keith-zephyr "remove myself" can provide signals for specific collaborators that aren't meeting the project role.  It also allows people to mark themselves as unavailable (on vacation for instance).  @nashif - This is fine, but we may get complaints from collaborators/maintainers that they didn't get a chance to review.
@keith-zephyr assignee is supposed to the gate preventing PRs from merging.  @nashif - not all PRs have an assignee.
@henrikbrixandersen - requiring that an assignee be gate keeper if they know a specific collaborator needs to review would complicate the process too much.
@nashif - we are improving with this.  Only 16% of PRs merged without approval of assignee.  @nashif - PR stats track assignees that are not managing their assigned PRs
@dleach02 - has assignee he merged PRs when the PR met the other criteria without approving himself
@carlescufi - he sometimes pings the assignee to ask them directly if they are okay with the PR merging.  In this case the assignee didn't have time to do a thorough review personally, but are okay with the general direction and the approvals from others.
@nashif - we need more consistency on the PR merge criteria

fabiobaltieri, 2024-01-10:
Few people in #release pointed out that GitHub has a `review_request_removed` event as part of the issue event APIs so this should be feasible.

marc-hb, 2024-01-10:
> why are collaborators not reviewing PRs. Why would they need to remove themself from a PR?

There can be many reasons, good ones and bad ones. Both types of examples have already been mentioned above. I suspect the most common "good" reason is that no one is interested in micromanaging per-file rules. In fact there's no reason for different parts of the same file to all have the same expert - good luck micro-managing that! So in practice just assign the whole team to the whole subdirectory and move on with your life.

Even when a collaborator opts out of some PRs for some "bad" reason, enforcing notifications is never going to enforce _participation_. If nothing else it will backfire and make them ignore GitHub notifications even more; including some notifications they did not want to miss. Notification spam and automated bots will never help with _that_ type of maintainership problem. Some problems can only be solved by humans.

PR #67457 (2024-01-11): scripts: set_maintainer: do not re-add self-removed reviewers
If a collaborator removes themselves from the reviewer list, do not
attempt to re-add them on changes to the PR.

Fixes #67214
commit 0d7d39d4 (2024-01-11): scripts: set_maintainer: do not re-add self-removed reviewers

If a collaborator removes themselves from the reviewer list, do not
attempt to re-add them on changes to the PR.

Fixes #67214
closed 2024-01-11 by dleach02