zephyrproject-rtos/zephyr #67244: shields: listing of shields only shows shields in zephyr root

nordicjm, 2024-01-05:
**Describe the bug**
If an invalid shield is supplied a list of shields is printed out. However, this only shows the shields in zephyr itself, any shields in module directories are not printed.

**To Reproduce**
Build a sample with an invalid shield name having moved some shields to modules

**Expected behavior**
All shield names to be printed

**Impact**
Minor

**Environment (please complete the following information):**
 - Commit SHA or Version used: 7ef10d0f718dd65e38f80f48f1c3e199acef02e0

henrikbrixandersen, 2024-07-07:
@nordicjm Wasn't this fixed by https://github.com/zephyrproject-rtos/zephyr/pull/71335?

nordicjm, 2024-07-08:
> @nordicjm Wasn't this fixed by #71335?

That it has

closed 2024-07-08 by nordicjm