zephyrproject-rtos/zephyr #67250: MAINTAINERS.yml: Too many "reviewers"

alwa-nordic, 2024-01-05:
# Problem statement
At the weekly Zephyr Bluetooth meeting, we brought up an issue: The automatic assignment of reviewers by the Zephyr bot does not correspond with the reality of who is responsible for reviewing.

The Zephyr bot is requesting reviews from people who usually never respond. This contributes to diffusion of responsibility. It happens quite often that a PR 'falls between the chairs'; it has a bunch of reviewers assigned, none of which feel it's their area of responsibility to review that PR.

Conversely, it also happens that I am in the middle of a review (and intend to ask for changes) but then the PR is merged without my approval, even though I am in the reviewers list. This is a waste of my time.

Sometimes we explicitly want a specific person to do a review for us. In those cases, it would be perfect if this was signaled by adding them to the reviewers list. But since the list of reviewers is populated with inactive people, this does help us to keep track of this.

See also: https://github.com/zephyrproject-rtos/zephyr/issues/67214

# Proposal
**Add an `observers` field** in addition to `collaborators` in MAINTAINERS.yml for stakeholders who want to keep up-to-date and occasionally chime in. 

Users in `observers`
- will be notified. (The bot can add a comment with the usernames.)
- are not required to respond.
- may add themselves to the reviewers.

For example, someone from the network subsystem is interested in potentially breaking changes in Bluetooth behavior.

## Alternatives

1. Some bespoke solution for keeping track of required reviewers separate from the core GitHub UI.
2. Remove the inactive reviewers from the `collaborators` list. They can keep up-to-date by periodically browsing the PRs in our subsystem using a custom filter.

henrikbrixandersen, 2024-01-05:
> Remove the inactive reviewers from the collaborators list. They can keep up-to-date by periodically browsing the PRs in our subsystem using a custom filter.

From my point of view, this is the correct and best solution. No need for inventing something new when we already have an established role for collaborators, with responsibility for responding to reviews. 

Inactive users should have their roles suspended, whether they are collaborator, maintainer, release manager, ...

stephanosio, 2024-01-06:
> The automatic assignment of reviewers by the Zephyr bot does not correspond with the reality of who is responsible for reviewing.

The semantic of "reviewers" for a GitHub PR is more close to "review wanted" and not "review required."

> Sometimes we explicitly want a specific person to do a review for us.

That is what "assignee" is for, which is automatically set to the maintainer of the most relevant area for each PR, and the assignee(s) **must** approve the PR before it can be merged -- there is no need to introduce a new concept like "observer" for this purpose.

At most, we could further clarify and document that one may add themselves as an "assignee" for a PR if they want to ensure that the PR is not merged without their review.

> The Zephyr bot is requesting reviews from people who usually never respond. This contributes to diffusion of responsibility.

zephyrbot adds reviewers based on the maintainers and collaborators fields of the relevant areas sorted by the number of files affected for each area. If maintainers and collaborators of the relevant areas are unresponsive, we should look into ways for improving their engagement (e.g. pinging them more actively or replacing them), not introducing a new field such as "observer."

> It happens quite often that a PR 'falls between the chairs'; it has a bunch of reviewers assigned, none of which feel it's their area of responsibility to review that PR.

This situation is not going to change even if we introduce the "observer" field -- there will still be maintainers and collaborators from multiple areas in the reviewer list for the PRs that "fall between the chair," and none of them will feel obligated to review the PR.

It is ultimately up to the "assignee(s)" of the PR to manage the review process by providing a review and/or requesting review(s) from the maintainers and collaborators that they consider important to verifying the correctness of the PR.

thedjnK, 2024-01-06:
> From my point of view, this is the correct and best solution. No need for inventing something new when we already have an established role for collaborators, with responsibility for responding to [...]

To add to this, from the many dev review meetings I've been in and actually from PRs I've submitted a long time ago, it's one of the most frustrating things to have "maintainers" that do not maintain their area and mean that PRs go on for weeks, months or even years without being merged/reviewed because the maintainer does not respond to any comments or requests to review, I fully support the idea of removing people if they fail to meet the requirements of the role they have been assigned.

Thalley, 2024-01-09:
Adding the reference to https://docs.zephyrproject.org/latest/project/project_roles.html where the roles are defined. One of the related responsibilities for this discussion here would be the item under "Collaborator":

> Responsibility to review relevant code changes within reasonable time.

With regards to

> Inactive users should have their roles suspended, whether they are collaborator, maintainer, release manager, ...

I agree. It should partly (mostly?) be up to the maintainer(s) to handle the collaborators for the their area (including adding, removing, pinging, etc.), assuming that the maintainer is in fact active. 

I do, however, miss some way of subscribing to a subset of PRs/issue, which is what @alwa-nordic suggests here with the observer role.

fabiobaltieri, 2024-01-10:
> The automatic assignment of reviewers by the Zephyr bot does not correspond with the reality of who is responsible for reviewing.

Can you come up with a custom logic for Bluetooth that works better than what the bot does now? Maybe we can hook the area with a custom assignment logic rather than the default one.

> Sometimes we explicitly want a specific person to do a review for us

Maybe you could set everyone who you want a review for as assignee and do a request changes  until they all approved?

> the list of reviewers is populated with inactive people

Ok, are you sending a PR to drop them or you expect someone else to do it? Happy to do it for you if it helps, it's not that many names to check manually.

keith-zephyr, 2024-01-10:
Process WG AIs
* @dleach02 to look at prioritizing maintainers by the zephyrbot
* @nashif - to investigate updating assignee for maintainers file specifically  (long term project)

----
@dleach02 - zephyrbot assigns too many reviewers and it's often not right people
@nashif - smaller subsystems/areas in maintainers file is probably a better approach.
@dleach02 - when multiple areas are touched by a PR - the bot should be smarter to make sure maintainers are selected first.
@henrikbrixandersen - this is a good idea and should be doable with the github tooling.
@henrikbrixandersen - the maintainer file is maintained by Maureen - but the maintainer of the corresponding area doesn't get notified or isn't required to approve.
@nashif - We need a process to add collaborators/maintainers to ensure they agree to the project roles.  Generally, Maureen does add the maintainer of an area as reviewer.
@henrikbrixandersen a better process would be to change the assignee to maintainer
@henrikbrixandersen - we see a lot of self- assignees to maintainer file.  Would prefer only maintainers nominate new collaborators
@nashif - we can change the maintainer file rules to automatically set the assignee based off the area.
@keith-zephyr - does anyone one to champion the "observer" feature as proposed?

keith-zephyr, 2024-06-05:
@dleach02 started some work a few months back.  Will pick this back up.

keith-zephyr, 2024-06-05:
@dleach02 script will identify all the areas affected by the PR and try to assigned maintainers from each area are added as reviewers before filling up the queue with collaborators from a single area.
@nashif There is some load balancing, but it could be improved.

dleach02, 2024-06-06:
Here is my draft PR to try to be more fair in assigning reviewers.

[73873]