zephyrproject-rtos/zephyr #67257: drivers: voltage_divider_scale_dt does not handle negative voltages

goodsireric, 2024-01-05:
**Describe the bug**
In the file: include/zephyr/drivers/adc/voltage_divider.h

voltage_divider_scale_dt() fails when the measured voltage is negative, because of the uint32_t types of the full_ohms and output_ohms. Instead of outputting a negative number, the output is a large positive number. 

**To Reproduce**
In tests/drivers/adc/adc_rescale/scr/main.c
Add a negative value for the input_mv, such as -1

This was discovered while grounding a Nordic nrf52840 ADC pin and seeing occasional max voltage values pop up instead of the expected near zero readings. 

**Expected behavior**
Voltage is kept negative

**Impact**
None, it's easy to copy the conversion and typecast to int32_t to keep the sign

**Logs and console output**
N/A

**Environment (please complete the following information):**
N/A

**Additional context**
N/A

kartben, 2024-01-05:
@goodsireric thanks for the report! Maybe you'd be interested in submitting a pull request? :)

goodsireric, 2024-01-08:
Yes, I will do that!

closed 2024-03-24 by github-actions[bot]